pothosware/SoapySDRPlay3 #1: Compilation errors on Windows

vsonnier, 2020-04-11:
Hi @SDRplay ! I finally have (some) time to test the SoapySDRplay3 module under Windows10 x64.

I installed the API/DRIVER 3.06 package for Windows, side by side to the existing 2.13 package.

Unfortunatly, I quickly stumbled under the following errors:
- FindLibSDRPlay.cmake : for `WIN32`, the definition still references `mir_sdr.h` / `mir_sdr_api.lib` instead of the   `sdrplay_api.h` / `sdrplay_api.h`. No big deal, I changed that.
- Next step, the compilation failed because `HANDLE` was unknown. Indeed, its definition appeared nowhere in the include files.

My renewed interest for SDR is because I bought a Youloop from rtl-sdr.com :) I live in appartment, in a urban environnement, and it is the first time I can finally use my RSP2 on the HF band !

I plugged it in Port A. I also tried a horrible hack job to plug it in the Hi-Z port trying to follow the plug-your-coax-in-Hi-z method of the guide with no success but I'm terrible at that kind of things.
  
I got decodable LSB/USB signals eventually on the 40 m and 80 m band using port A. Since it is a passive loop, I'm forced to use it on RF ctrl = 0/1/2 and AGC on to be able to pick up signals, but the noise level was indeed much lower than when I used a simple wire on the Hi-Z. 

Stopping the digression I wondered if the Soapy3 version put some better internal settings, given all the discussion we had with @fventuri and I wanted to test it.

vsonnier, 2020-04-17:
@SDRplay  ping ?

SDRplay, 2020-04-17:
@vsonnier thanks for this. @fventuri are you able to pull Vincent's changes into the master branch?

fventuri, 2020-04-18:
Thanks Vincent for your changes.
I just pushed the changes for 'FindLibSDRplay.cmake' and  'Settings.cpp' to the master branch.

As per the changes to the include file 'sdrplay_api.h', it is probably up to Andy and SDRplay to review them, and address the issues with that include file.

Franco

SDRplay, 2020-04-18:
Sorry guys, I missed the changes you noted for sdrplay_api.h! With the release of SDRuno 1.4 we are preparing to release API 3.07, you can already get a version of that here: https://www.sdrplay.com/software/SDRplay_RSP_API-Windows-3.07.exe which should address the issues raised.

vsonnier, 2020-04-18:
Hi @SDRplay thanks for API v3.07, it works for me on Windows.

fventuri, 2020-04-18:
Thanks Andy and Vincent.
If I don't hear from you about this issue in the next few days, I'll go ahead and close it.

Franco

vsonnier, 2020-04-19:
> If I don't hear from you about this issue in the next few days, I'll go ahead and close it.

I'm fine with this, so far it seems to work well.

One minor issue I noticed while browsing `Settings.cpp` : apparently RSPDx management is missing in  `std::string SoapySDRPlay::getAntenna(const int direction, const size_t channel) const`.

fventuri, 2020-04-19:
Good catch Vincent.
I just created another issue - Missing RSPdx entries in getAntenna() (https://github.com/SDRplay/SoapySDRPlay/issues/2) to track this problem.

It should be fixed in master now.

Franco

fventuri, 2020-04-29:
Closing as done.

closed 2020-04-29 by fventuri