pothosware/SoapySDRPlay3 #9: findSDRPlay should not stop stream

fmoessbauer, 2020-05-30:
When the device is streaming and findSDRPlay is called (which might be due to various reasons, esp. when running under SoapyRemote), the stream is stopped.

Here, I propose to instead just ignore the call and return no devices:

https://github.com/SDRplay/SoapySDRPlay/blob/master/Registration.cpp#L76

Patch (feel free to change, if accepted):

```
diff --git a/Registration.cpp b/Registration.cpp
index db02814..71c6a33 100644
--- a/Registration.cpp
+++ b/Registration.cpp
@@ -73,8 +73,8 @@ static std::vector<SoapySDR::Kwargs> findSDRPlay(const SoapySDR::Kwargs &args)
 
    if (activeStream)
    {
-      SoapySDR_log(SOAPY_SDR_WARNING, "findSDRPlay() called while the device is streaming. Deactivating stream.");
-      activeSoapySDRPlay->deactivateStream(activeStream, 0, 0LL);
+      SoapySDR_log(SOAPY_SDR_WARNING, "findSDRPlay() called while the device is streaming. Ignoring request.");
+      return {};
    }
 
    if (deviceSelected)
```

fventuri, 2020-05-30:
@fmoessbauer - Thanks Felix for reporting this issue. I just fixed it as you suggested.
Please give it a try and let me know.

Franco

fmoessbauer, 2020-06-01:
Hi @fventuri , just tested the latest master and my use-case now works. Thanks!

closed 2020-06-01 by fmoessbauer