pothosware/SoapySDRPlay3 #12: Problem setting frequency with RSPduo

righthalfplane, 2020-06-24:
I have been working on getting SdrGlut to work with the RSPduo. I have managed to open the two channels on one device in "Dual Tuner" mode. Almost everything is working - except when I change the center frequency - both channels are effected. I looked at the code and "setFrequency" is ignoring the channel number. If setFrequency would actually set the frequency for only the requested channel  - the program would be quite usable.

fventuri, 2020-06-25:
Thanks for letting me know about this issue @righthalfplane - my idea when I wrote the implementation of that SoapySDR driver was that people would use the 'Dual Tuner' mode mainly for diversity reception, i.e. with both tuners tuned to the same center frequency, but using different antennas, to take advantage of the different phase of the two incoming I/Q streams to cancel some of the noise.

This said, I just checked what SDRuno (which is what I use as a reference) does, and I see it has three sub-modes: 'DUAL (NORMAL)', 'DUAL (ADS-B)', and 'DIVERSITY', so perhaps in the SoapySDRPlay driver I should rename the current 'Dual Tuner' mode as 'DIVERSITY', and add another sub-mode called 'DUAL (NORMAL)' or something like that, and in that sub-mode keep the two tuners completely independent (and you would be able to set the center frequencies the way you want, based on the requested channel).
@SDRplay , what do you think?

By the way I didn't know SdrGlut up until tonight - I was able to compile it and run it here on my Fedora Linux using 'makefileUbunta', and it looks really nice! I haven't tried it much, but I plan to this weekend.
Congratulations on writing it!

Franco

righthalfplane, 2020-06-26:
I would be great to have both the DIVERSITY and the independent tuner modes. I completely forgot about the idea of diversity reception - it seemed strange to tune both streams to the same frequency.

fventuri, 2020-06-27:
@righthalfplane , I just created a new branch called 'dual-tuner-submodes' (https://github.com/SDRplay/SoapySDRPlay/tree/dual-tuner-submodes), that should support both diversity reception and independent RX for dual tuner mode.

I ran a couple of quick tests using CubicSDR, atried both submodes, and it ran without crashes (but CubicSDR has only one channel, so I am not sure if the second channel works as expected).

Please give it a try and let me know if it works for you.
Franco

righthalfplane, 2020-06-27:
I tried it but it did not work.  It tunes them both to the channel 0 frequency, when channel 0 is set. It tunes neither channel, when channel 1 is set.

I a few days ago forked your original version and got it to sort of work -

https://github.com/righthalfplane/SoapySDRPlay

by setting the tuner to sdrplay_api_Tuner_A for channel 0 and letting it set both channels for channel 1.  I had my program make one call for channel 0 and two calls for channel 1 - one to set channel 1 (and channel 0) followed by a call to reset channel 0 to the correct frequency.

fventuri, 2020-06-27:
@righthalfplane , I think I found the problem.

My code was calling `sdrplay_api_Update(device.dev, device.tuner, sdrplay_api_Update_Tuner_Frf, sdrplay_api_Update_Ext1_None)` with device.tuner set to both tuners.
I just changed it to update the center frequency only for the correct tuner (TunerA for channel 0, and TunerB for channel 1 when in independent RX mode).

Please try again with this fix and let me know how it goes.

Franco

righthalfplane, 2020-06-27:
That got it - It woks now - Thanks

fventuri, 2020-06-28:
@righthalfplane - Very good!

@SDRplay - Andy, please let me know what you think about adding the option to run the RSPduo dual tuner mode either in diversity reception mode (same center frequency), or as two independent RX (with different center frequencies).
If it looks good to you, I'll merge that branch into master.

Franco

righthalfplane, 2020-06-28:
For the "two independent RX" case -  is there any reason why the sample rate needs to be limited to 2e6 ?
The "gain" of both channels seems to be tied to one other and I suppose that the "gainmode" is also tied.

fventuri, 2020-06-28:
Dale,
in dual tuner mode (and also in master or slave modes) the sample rate that the client application sees from the RSPduo is always 2MHz - see this comment from Andy with SDRplay https://github.com/pothosware/SoapySDRPlay/issues/62#issuecomment-569296059 (you may want to go through the whole discussion thread there to read all the details). The sample rate can be further lowered with decimation (that's why the selection also offers 1MHz, 500kHz, and so on), but you can't make it any higher than 2MHz.

As per the gain and AGC settings, I think you have a valid point, and they should be made independent too.
I just made that change to the code in the 'dual-tuner-submodes' branch - give it a try and let me know.

Franco

righthalfplane, 2020-06-29:
Hi Franco,

Your fixes on the gains works great. Thanks!

Over the last few days, I have fixed many bugs in SdrGlut - so if you want to try it - pull the new changes.

Dale

fventuri, 2020-06-29:
Very good Dale!

I just re-cloned the github repo for SdrGlut and ran it for a few minutes here, and it is a very nice program! Unfortunately here at my desk I can only use a couple of plain wires as antennas, so I can only receive very few stations because of the RFI.

I also watched the very interesting YouTube video you made a few days ago using the RSPduo in dual tuner mode with SdrGlut, and I am very glad I found out about it (so far I've only used CubicSDR and gqrx, and I am working on the GnuRadio 3.8/3.9 OOT module these days).

Franco

pabs77, 2021-04-28:
Hey guys ! first of all thx to the collaboratos for this awesome work! I'm having issues trying to set my RSPduo to its dual mode using SDRsoapy. In SDRuno or console it works properly. Also --find shows me all four of its modes but I cannot set the wanted mode up. I have been reading the API and the documentation but cannot find the function/s to do so. Also it is set to Single tunner mode (I guess by default).  I hope you can help me.

fventuri, 2021-04-28:
@pabs77 
First of all thanks for the nice words.

The code that allows you to use the RSPduo in dual tuner mode as either a 'diversity receiver' (i.e. where the two RX are always tuned to the same frequency) and as independent receivers (each tuner has its own frequency) is still in the `dual-tuner-submodes` branch from about a year ago.
If you pull that branch and build it, you should see five different modes for your RSPduo (dual tuner mode should show you the two 'submodes' mentioned above); when set to 'independent receivers' you should be able to call setFrequency() on the two different channels (0 and 1), and set the two tuners frequencies independent of each other.

Please give it a try; let me know if you have any problems at all with it, and if it works for you and it does what you want, I'll update it to get it ready to be later merged with master.

Franco

pabs77, 2021-04-28:
Hey Franco! Thx a lot. I'll give it a try and will come back to you ! thx for the quick answer !

pabs77, 2021-04-28:
Sorry I'm kind of new to GitHub and tu "pull", "build" stuff ... can you please explain me how to use this branch into my code? I've been tryin g to build the .cpp files and to include the .hpp to the inc folder ..but it is not working. If you coul explain me how this works I'll be really greatful. I'm a student working on my thesis and I'm quite desperate for this to work. Sorry to bother you !

pabs77, 2021-04-28:
Well I got the .dll (just one file right) ?  and I shozuld include it into my module 0.8 folder with the other dlls right ? I'm trying now to use the functions specified in the sdrplay support dll but nothing seems to work. maybe you can enlighten me a little bit ? maybe a small example on how to call the changeRspDuoMode() because i cannot get it to work using my python scrypt ...thanks in advance

fventuri, 2021-04-29:
@pabs77 - thanks for your comments.
I apologize for the delay in answering your questions; I work my regular job during the day so I can only reply here either in the morning before work or after work/dinner like now (I live in Florida, in the East Coast of the US).

Anyway, the git commands to work on a branch other than master are:
```
git clone https://github.com/pothosware/SoapySDRPlay3.git
cd SoapySDRPlay3
git checkout dual-tuner-submodes
```
After that, you should be able to run the usual build commands (`mkdir build`, `cmake`, `make`, etc) as usual. Since you are running these commands under the `dual-tuner-submodes` branch, the code you will be compiling, installing, etc is the code from that branch.
If you need further help about git commands, a good resource is this git book available online for free: https://git-scm.com/book/en/v2

As per the other question you asked, I am running Linux here, so I am not really familiar with Windows and DLL files, but I imagine that with SoapySDR on Windows the API and the function calls are the same as in Linux, so you may want to start reading here: https://github.com/pothosware/SoapySDR/wiki; also this example should help you with your Python script: https://github.com/pothosware/SoapySDR/wiki/PythonSupport

Franco

pabs77, 2021-04-29:
Hey Franco! 
Sorry to bother, but nothing seems to work ! is there anybody who has achieved using the submodes branche in Windows ?? If so please let me know . I need some help. 

I get the sdrplaySupport.dll after building with cmake. After that I have tried to include it into the libraries path and also system variables path. All functions from the  SoapySDRPlay.hpp like getHardwareInfo() or setFrequency(), writeSettings() etc seem to work perfectly. But when I try to call the chageRspDuoMode()  function it tells me that device has not this module. What I'm doing wrong ? Im working with Python 3.9 and Visual Studio 2019... Everithing else works perfectly. But in single tunner mode ... I need the dual mode :( 
Is there an example of how to call the change mode function ?

sdr = SoapySDR.Device(dict(driver="sdrplay"))  # Create SDRplay instance
sdr.setSampleRate(SOAPY_SDR_RX, rx_chan, fs)          # Set sample rate
sdr.setGainMode(SOAPY_SDR_RX, rx_chan, use_agc)       # Set the gain mode
sdr.setFrequency(SOAPY_SDR_RX, rx_chan, freq)         # Tune the LO

print(sdr.getHardwareKey())
print(sdr.getHardwareInfo())
sdr.changeRspDuoMode()      #(doesnt recognize this function) 

Thx in advance !

fventuri, 2021-04-29:
@pabs77 
SoapySDR comes with a very handy utility called `SoapySDRUtil`; before you start working on the Python code, please run `SoapySDRUtil --find=driver=sdrplay` to make sure your module is built correctly and works properly. The output should be very similar to this:
```
SoapySDRUtil --find=driver=sdrplay

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  driver = sdrplay
  label = SDRplay Dev0 RSPduo 1234567890 - Single Tuner
  rspduo_mode = 1

Found device 1
  driver = sdrplay
  label = SDRplay Dev1 RSPduo 1234567890 - Dual Tuner (diversity reception)
  rspduo_mode = 2

Found device 2
  driver = sdrplay
  label = SDRplay Dev2 RSPduo 1234567890 - Dual Tuner (independent RX)
  rspduo_dual_tuner_independent_rx = true
  rspduo_mode = 2

Found device 3
  driver = sdrplay
  label = SDRplay Dev3 RSPduo 1234567890 - Master
  rspduo_mode = 4

Found device 4
  driver = sdrplay
  label = SDRplay Dev4 RSPduo 1234567890 - Master (RSPduo sample rate=8Mhz)
  rspduo_mode = 4
  rspduo_sample_freq = 8
```

If the output looks good, then you should be able to use your RSPduo in dual tuner mode with independent RX with something like this (I just modified the code in the example here:https://github.com/pothosware/SoapySDR/wiki/PythonSupport to give you a rough idea):
```
import SoapySDR
from SoapySDR import * #SOAPY_SDR_ constants
import numpy #use numpy for buffers

#enumerate devices
results = SoapySDR.Device.enumerate()
for result in results: print(result)

#create device instance
#args can be user defined or from the enumeration result
args = dict(driver="sdrplay", rspduo_mode="2", rspduo_dual_tuner_independent_rx="true")
sdr = SoapySDR.Device(args)

...

sdr.setFrequency(SOAPY_SDR_RX, 0, 912.3e6)     # sets the frequency for the first RX

...

sdr.setFrequency(SOAPY_SDR_RX, 1, 89.9e6)      # sets the frequency for the second RX

...

```

Hope it helps,
Franco

pabs77, 2021-04-29:
Hey Franco...
 It didnt help!  it solved it ! Thx a lot man ! the problem was how  I was passing the args. I learned a lot thx to you and this posts ! thx man ! have a ncie one ! if you need me to explain how to get the .dll libraries to work in Windows just let me know ... 
Thx again for the nice work ! 
Have a nice day ! 
Grettings from Germany !

fventuri, 2021-04-29:
@pabs77 
I am really glad to hear that it works for you.

Good luck with your project!
Franco

closed 2021-04-29 by fventuri