pothosware/SoapySDRPlay3 #13: Problems building SoapySDRplay on Windows 7

righthalfplane, 2020-06-26:
I tried to build SoapSDRPlay on Windows 7. It complained about a missing definition of "HANDLE" - I put in the version of "HANDLE" out of "phread.h" and that got it to complain about "sdrplay_api_DbgLvl_Disable" missing. I commented out the lines with the sdrplay_api_DebugEnable calls.  It then built and ran in "Duel Turner" mode.

fventuri, 2020-06-26:
@righthalfplane , I work on Linux most of the time, therefore I am not a Windows expert by any stretch of imagination, however looking at the comments by @vsonnier and @SDRplay on this very similar issue: https://github.com/SDRplay/SoapySDRPlay/issues/1, it seems like both of your issues should be fixed in SDRplay Windows API 3.07, which you should be able to download using the link provided in the other thread (https://www.sdrplay.com/software/SDRplay_RSP_API-Windows-3.07.exe).

Hope this helps,
Franco

righthalfplane, 2020-06-27:
Thanks - That fixed it.

closed 2020-06-27 by fventuri