pothosware/SoapySDRPlay3 #17: rspduo in diversity mode (mode 2)

bob-uk, 2020-11-20:
I am working to get the rspduo working in mode 2 (diversity) feeding into gnu radio via the soapy interface.  I have modified Streaming.cpp (probably inelegantly due to rusty c++!) to get both channels back to gnu radio (locally and remote).  It appears to be working but the data doesn't appear to be synchronised, I think because the phase difference between the two channels varies considerably over short time spans.  Modified Streaming.cpp attached, please no laughter😉. My soapy module in gnu radio is configured for 2 channels and sdrplay with a config line of "remote=192.168.0.10:1234,driver=remote,rspduo_mode=2".  Any ideas about the data synchronisation? Thanks Bob
[Streaming.cpp.zip](https://github.com/SDRplay/SoapySDRPlay/files/5575820/Streaming.cpp.zip)

fventuri, 2020-11-21:
Bob,
thanks for your input.

Since you say you use GNU Radio to connect to the SDRplay RSPduo, I am wondering if you looked into the GNU Radio OOT module for the SDRplay RSPs:
- for GNU Radio version 3.7, the OOT module can be found here: https://github.com/fventuri/gr-sdrplay/tree/API3+RSPduo (note the branch name 'API3+RSPduo')
- for GNU Radio version 3.9 (and above), please see here: https://github.com/fventuri/gr-sdrplay3

I am not familiar with your use case, but using the OOT modules (instead of passing through the SoapySDR interface first, and then the SoapySDRPlay driver) might help with your issue.

Franco

bob-uk, 2020-11-21:
Franco
Thanks for getting back to me.
I have made progress (I think) with the SoapySDRPlay code and revised files attached.  I think I still have a problem between rx_callback and receiveStream but it is mainly in sync.
[SoapySDRPlay.zip](https://github.com/SDRplay/SoapySDRPlay/files/5578670/SoapySDRPlay.zip)

On the issue of the gr-sdrplay OOT modules, I hadn't clocked their import so am looking at them now. As I said I have gr 3.8 loaded and running so have built 3.9 and installed it to a diffefent location, hopefully they won't trample on each other.  I am running on mac Catalina and have some problems compiling gr-sdrplay3, both with clang and gcc/g++

in rsp_impl.h this construct fails "const double (&get_freq_range() const override)[2];" with a compiler error requesting a ')' before the override?
and then it is failing deep in the Xcode includes
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/type_traits:1547:38: error: 
      implicit instantiation of undefined template
      'std::__1::hash<sdrplay_api_ReasonForUpdateT>'
    : public integral_constant<bool, __is_empty(_Tp)> {};

so a bit stuck on this route at present

Bob

bob-uk, 2020-11-21:
Sorry Franco, my bad didn't mean to close this, I am a novice in github

fventuri, 2020-11-22:
Bob,
I'll look at your changes to SoapySDRPlay tomorrow morning.
As per the issues compiling the GNU Radio OOT module gr-sdrplay3, I created a new issue here: https://github.com/fventuri/gr-sdrplay3/issues/1 (see my answer there).

Franco