pothosware/SoapySDRPlay3 #20: Error probing device: Can't find label in args

ReneeVandervelde, 2020-12-02:
Getting an error I haven't seen reported anywhere when trying to use this:

```shell
$ SoapySDRUtil --probe="driver=sdrplay"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay
[WARNING] Can't find label in args
Error probing device: Can't find label in args
```

Could this be due to a recent change or version conflict with one of the libraries?

Here's the info:

```shell
$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-g926c86d9
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib64/SoapySDR/modules0.8
Module found: /usr/local/lib64/SoapySDR/modules0.8/libremoteSupport.so  (0.6.0-c09b2f1)
Module found: /usr/local/lib64/SoapySDR/modules0.8/librtlsdrSupport.so  (0.3.1-52cb5c0)
Module found: /usr/local/lib64/SoapySDR/modules0.8/libsdrPlaySupport.so (0.3.0-5c7c21f)
Available factories... remote, rtlsdr, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

on an arm64/aarch linux system.

ReneeVandervelde, 2020-12-02:
Interesting. This issue went away with a reboot. Thinking the device might have been in use from another tool I was trying?
Unsure, could maybe use a better message, if that's the case.

closed 2020-12-02 by ReneeVandervelde

fventuri, 2020-12-03:
Good point Renee.

The times I saw a message like yours were when the OS was not able to 'see' the RSP as a USB device for some reason.

I added this diagnostic info to the error message you saw, and I also added a 'Troubleshooting' section to the README explaining what to do if that message appears.

Franco