pothosware/SoapySDRPlay3 #23: implement device filter and open by serial

guruofquality, 2020-12-28:
As part of standards compliance (just kidding). The find function and open should support the "serial" keyword, filtering on the keyword when specified in the hints, and even using it to open the device in the factory function. But most of the drivers actually support this so you can almost universally pass in a serial number in the args to uniquely specify a device and it should actually simplify the code a bit:

Here are a few notes:

**SDRPlay v2 has a good example of this**

* Take a look at v2 https://github.com/pothosware/SoapySDRPlay2/blob/master/Registration.cpp
* and https://github.com/pothosware/SoapySDRPlay2/blob/master/Settings.cpp#L45

**Implementation changes**

* Dont parse the label string in Settings.cpp, just use the args["serial"] directly to match the device
* In Registration.cpp, just add dev["serial"]  = format(... SDRPLAY_MAX_SER_NO_LEN, rspDevs[i].SerNo)
* And in Registration.cpp, skip the loop if dev["serial"] != args["serial"] when serial is present in the args

**Other notes**

* `dev["driver"] = "sdrplay";` is redundant, SoapySDR library will handle that

fventuri, 2020-12-28:
Thanks Josh for the useful feedback.
I'll work on making those changes.

Franco

fventuri, 2020-12-28:
@guruofquality - the reason for the convoluted (and ugly) code in Registration.cpp (and Setting.cpp) is because the RSPduo can have several 'modes of operation' (Single Tuner, Dual Tuner, Master, Slave) as  described in the SDRplay API specification (https://www.sdrplay.com/docs/SDRplay_API_Specification_v3.06.pdf). The mode of the operation for the RSPduo must be known (i.e. chosen by the user) before the device is selected with the `sdrplay_api_SelectDevice()` API call, which happens in the constructor (https://github.com/pothosware/SoapySDRPlay3/blob/master/Settings.cpp#L180).

A while ago Andy (@SDRplay ), Vincent (@vsonnier ) and I had a long discussion about how to best handle these multiple RSPduo modes (see here: https://github.com/pothosware/SoapySDRPlay2/issues/62 and here for a previous discussion: https://github.com/pothosware/SoapySDRPlay2/issues/58), and the solution we adopted was to have multiple 'pseudo-devices' for the same RSPduo, one for each of the possible modes (for instance single tuner with tuner A, single tuner with tuner B, dual tuner, master with tuner A, master with tuner B, or just slave with the available tuner in case another application is running in master mode).
This approach gave us the benefit that at creation time we know which one of the various RSPduo configuration modes the user want to run, but has the drawback that the serial number alone is not enough to know how to the RSPduo should be configured, i.e. we do not have enough information with just the SN to be able to run the constructor.

Since you are more familiar than me on how other SoapySDR drivers work, perhaps you might have already encountered a similar scenario, and therefore know a better option that would still allow the user to choose among the various mode of operation that the RSPduo model offers.

Franco

guruofquality, 2020-12-29:
Just for the sake of argument, I think pseudo-devices can be a good solution and its certainly nice for applications like cubicsdr or gqrx that give nice drop downs with the label. With pseudo-devices, I would still do everything the same with the serial number as discussed, but also add a keyword like "mode".

* If the find function passed in a "mode" string, it would eliminate options that did not match the "mode"
* And in the constructor, "mode" would be used to  sdrplay_api_SelectDevice()
* And of course if the user passed in a serial, you still get the desired behaviour of only selecting the SDRPlay of interest. Or lets say its an RTLSDR serial, you would eliminate SDRPlays from being discovered, which is one of the ways we can "play nice" with other drivers by supporting the "serial" convention

fventuri, 2020-12-30:
@guruofquality - I just pushed the changes you suggested to the branch 'support_for_serial_keyword' (https://github.com/pothosware/SoapySDRPlay3/tree/support_for_serial_keyword).

I just ran a couple of quick tests here with the RSPduo; I was able to run CubicSDR in Single Tuner mode (and Dual Tuner mode), and two instances of CubicSDR at the same time, one in Master mode, and the other in Slave mode, so the basic functionality should work.

Franco

guruofquality, 2020-12-30:
looks good

fventuri, 2020-12-31:
Thanks @guruofquality for checking.
I found (and fixed) another problem with the variable `outputSampleRate` not being initialized, that was causing gqrx to crash at startup.
I pushed the code change to the `support_for_serial_keyword` branch a few minutes ago; give it a try again and let me know if you see any other problem.

@nmaster2042 - if you have some time, can you try running the code from this new branch `support_for_serial_keyword` with gqrx (and possibly CubicSDR) to make sure that everything works as expected?

Thanks,
Franco

nmaster2042, 2021-01-01:
Happy new year to you all.

@fventuri OK, I'm starting tests with the new support_for_serial_keyword.

I'll give results here ASAP.

nmaster2042, 2021-01-02:
Hi @fventuri 

I made test with RSP2, gqrx 2.14.3, gr-osmosdr official and new support_for_serial_keyword branch.

I can't use RSP2, because I get an error, see below.

[image]

I didn't get this with master branch you updated some days ago with string args.

Also tried with CubicSDR, no issue here.

fventuri, 2021-01-02:
Good catch @nmaster2042 - I had forgotten the initialization of the variable `outputSampleRate` for the non-RSPduo devices.
I just fixed that problem in the branch `support_for_serial_keyword`; please try again with the latest code, and let me know how it goes.

I'll also run some more tests here with switching RSPduo modes and tuners/antennas while the RSPduo is streaming to make sure the client application (CubicSDR or gqrx) does not hang, like it used to when I started working on this driver.

Franco

nmaster2042, 2021-01-03:
Hi @fventuri I made tests on your new commit.

It's now working as expected with gqrx, cubic, and sdrplusplus, a new SDR software in Beta I'm testing right now.
On Gqrx, no more crash, we can see now serial number on selected device, for duo, the mode is now a string.
String args still working like the last update you made on mater branch in gqrx.

So it seems to work fine.

Thank you for the great work.

fventuri, 2021-01-03:
Thanks for your very extensive tests @nmaster2042

Yesterday afternoon/evening I played around a bit with CubicSDR and the RSPduo, and I noticed that trying to change the RSPduo mode from say Single Tuner to Master while it is streaming causes an exception to be thrown.

I also noticed that the code for the cached results for claimed handles (that I copied almost exactly from the SoapySDRPlay2 driver) does not to seem to work correctly for the different RSPduo modes; according to the change log for the SoapySDRPlay2 driver release 0.3.0 (https://github.com/pothosware/SoapySDRPlay2/blob/master/Changelog.txt#L5), this change should make things work with SoapySDR server, so I suspect more tests are needed with SoapySDR server, to make it work as it should with the RSPduo.

Franco

nmaster2042, 2021-01-03:
OK @fventuri.

Until now, I mainly made tests with devices attached to my computer, except for RSP DX that is used remotely with a SoapySDRServer.

I will swap RSP Duo and Dx, to try duo modes remotely.

With Gqrx, I have issues with SoapySDRServer, crash for example when I change demodulator.
I'll chaek with a non RSP device to see if it's a gqrx issue.

With gqrx, it's always more complex to detect the layer causing issues (gqrx, gr-osmosdr, soapy).

I'll post results here ASAP.

nmaster2042, 2021-01-04:
@fventuri , here is my test results with RSP Duo on SoapySDRServer, for now only single tuner mode.

Single tuner mode seemed to work initially.

[image]

Start Cubic, select device, start streaming.

[image]

Then I stopped streaming, refresh device list and select RSP duo single tuner mode, same antenna B and Bias T.

[image]

And start streaming again.

Here it seems antenna port B is not selected. I had to select Antenna A then Antenna B to get back to whate I listenet before.

[image]

Other issue, Cubic on RSP duo, single tuner mode, I started a second instance of Cubic, to check what will happen in my device list.

[image]

And I see I have RSP Duo - Single Tuner, I can't get option settings on the right of the window.
I think as I'm runing a first instance in single tuner mode I shouldn't see the RSP Duo in the second Cubic, right ?

I'll continue with other modes.

nmaster2042, 2021-01-04:
Second test set with RSP duo on SoapySDRServer, Dual Tuner mode.

I don't know really what this mode can be used for (diversity ?).

In First instance of Cubic, selecting RSP duo - Dual Tuner mode and start streaming is OK.

On the second instance of Cubic, like in Single Tuer mode:

[image]

If dual tuner mode is already used, I think it shouldn't be displayed in the device list.

nmaster2042, 2021-01-04:
Now this is the test set for master mode on RSP duo, SoapySDRServer, Cubic SDR.

First, Start of Cubic, select the remote RSP duo - Master, set Tuner 2 + Bias T.

[image]

Then Start Streaming, all is fine.

[image]

Stop stream, open device list, refresh device list, keep same setup (Tuner 2 + Bias T).

[image]

Seems like not the right tuner selected. At this stage I need to close and restart Cubic to get things working as espected.

[image]

Test set for slave mode coming.

nmaster2042, 2021-01-04:
Last test set, Slave mode RSP duo on SoapySDRServer while a Cubic instance is using RSP duo Master mode, Tuner 2 + Bias T.

Start of the second instance of cubic, I get 2 Rsp duo - Slave avaliable

[image]

On the 2 device I cat see options.

[image]

Choosing the first one and start streaming, OK.

[image]

Stop streaming, go to device list, refresh, I get now only one slave device.

[image]

Start stream again, and it works, master continues to stream on first cubic instance.

Now on the first instance of cubic (master) I stop streaming, slave continues to stream on second instance.

In first instance stopped master, I go to device list, refresh, I get a slave device only.

[image]

But can't sekect and start because Cubic crashes.

[image]

Something failed, see SoapySDRServer log screen below:

[image]

The second instance of cubic (slave) freeze too:

[image]

I had to kill this instance of cubic by myself.

[image]

I hope this will help you do identify what is to be modified in the driver.

fventuri, 2021-01-05:
Thanks for all your extensive tests and screenshots @nmaster2042 !

Tonight I just looked at one of the problems you found with SoapySDRServer - namely the fact that if you run two instances of CubicSDR, and the first one is say in Single Tuner mode, the second instance too sees the RSPduo in Single Tuner mode (going through SoapySDRServer, of course).

At first I thought too it was wrong that the second instance of CubicSDR would also see the RSPduo, then I found out this issue https://github.com/pothosware/SoapySDRPlay2/issues/46 for the old version of the SoapySDRPlay driver (i.e. SoapySDRPlay2); after reading it, I understood that that behavior is by design (see the changes by @guruofquality in this pull request https://github.com/pothosware/SoapySDRPlay2/pull/47), and therefore should be replicated in the new SoapySDRPlay3 driver.

In order to make sure that it works exactly the same way, I ran these two 'SoapySDRUtil' commands in two different terminal windows as suggested by @fmoessbauer (https://github.com/pothosware/SoapySDRPlay2/issues/46#issuecomment-457113648):
```
SoapySDRUtil --args="driver=remote,remote=127.0.0.1,remote:driver=sdrplay,remote:prot=tcp" --rate=1e6 --direction=RX
```
and:
```
SoapySDRUtil --find="driver=remote,remote=127.0.0.1,remote:driver=sdrplay"
```

Since I noticed that, in the case of the RSPduo, the serial number is not enough to uniquely identify the selected RSPduo mode ('Single Tuner', 'Dual Tuner, 'Master', etc), I made a few changes to the code in the `support_for_serial_keyword` branch to add the RSPduo mode to the claimed serial cache key.

Tomorrow after work I'll start looking at the problems with the crash and freeze of CubicSDR (which should never happen).

Franco

nmaster2042, 2021-01-05:
OK @fventuri I saw your changes for RSP duo mode to the claimed serial cache key.

On Cubic now if I start a master mode stream, on a second instance device list I can see 2 RSPduo devices: Master and Slave.

fventuri, 2021-01-05:
@nmaster2042 - that is to be expected on the second instance of CubicSDR when the first instance is running in Master mode:

- the Slave instance comes from the fact that the RSPduo has its slave "half" available for streaming (as per the Master/Slave architecture in the RSPduo), and if you click on that, you should be able to stream in slave mode, as usual
- the Master instance comes from the "claimed serial cache" inside SoapySDRServer (as discussed here: https://github.com/pothosware/SoapySDRPlay2/issues/46); I implemented that logic the same way that @guruofquality did in his commit, however to be very honest, I am not sure what purpose it serves yet (I need to re-read that discussion tonight to understand better).

Franco

fventuri, 2021-01-10:
@nmaster2042 , @pausrn - I just pushed several changes to the `support_for_serial_keyword` branch (https://github.com/pothosware/SoapySDRPlay3/tree/support_for_serial_keyword), that should hopefully address many problems with multiple clients and SoapyRemote.

These are the tests I ran here with my RSPduo; they all ran without any crash or freezing the client application (in all these tests I started SoapyRemote in a different terminal with the command: `SoapySDRServer --bind`):

- First test (Single Tuner mode):
client 1:
```
SoapySDRUtil --args="driver=remote,remote=127.0.0.1,remote:driver=sdrplay,remote:prot=tcp" --rate=1e6 --direction=RX
```

client 2:
```
SoapySDRUtil --args="driver=remote,remote=127.0.0.1,remote:driver=sdrplay,remote:prot=tcp" --rate=1e6 --direction=RX
```

- Second test (Master mode):
client 1:
```
SoapySDRUtil --args="driver=remote,remote=127.0.0.1,remote:driver=sdrplay,remote:prot=tcp,remote:mode=MA" --rate=1e6 --direction=RX
```

client 2:
```
SoapySDRUtil --args="driver=remote,remote=127.0.0.1,remote:driver=sdrplay,remote:prot=tcp,remote:mode=MA" --rate=1e6 --direction=RX
```

- Third test (Master + Slave mode):
client 1:
```
SoapySDRUtil --args="driver=remote,remote=127.0.0.1,remote:driver=sdrplay,remote:prot=tcp,remote:mode=MA" --rate=1e6 --direction=RX
```

client 2:
```
SoapySDRUtil --args="driver=remote,remote=127.0.0.1,remote:driver=sdrplay,remote:prot=tcp,remote:mode=SL" --rate=1e6 --direction=RX
```

Please run your tests with the new code, and let me know if you experience any problem (crashes, freezes, or others) - if you can reproduce the problem with a CLI command like `SoapySDRUtil` or `rx_tools`, please post the exact command, since it makes troubleshooting the problem easier for me.

Franco

pausrn, 2021-01-10:
Tried with SoapySDRUtil, cubicSDR and rx_tools. Only problem I have seen is that, except for the first , starting a stream with a sample rate greater-than or equal to 3Msps will make the server crash with a segfault : 

Ex : 
Starting a first stream (at any sample rate)
`SoapySDRUtil --args="driver=remote,remote=127.0.0.1,remote:driver=sdrplay,remote:prot=tcp" --rate=6e6 --direction=RX`

Starting another stream (at less than 3Msps)
`SoapySDRUtil --args="driver=remote,remote=127.0.0.1,remote:driver=sdrplay,remote:prot=tcp" --rate=2e6 --direction=RX`

Starting another stream (at more than 3Msps)
`SoapySDRUtil --args="driver=remote,remote=127.0.0.1,remote:driver=sdrplay,remote:prot=tcp" --rate=3e6 --direction=RX`

Here the server will crash with a segfault

This can be replicated with rx_tools and cubicSDR :
Starting iq recording with rx_sdr
`./rx_sdr -f 1700M -s 6M -d "driver=remote,remote=tcp://[::]:55132" -I CS16 -F CS16 test_METEOR.raw`
then starting CubicSDR with a sample rate higher than 3Msps will crash the server

The same happen with two CubicSDR instances, two rx_tools instances, one rx_tools and one SoapySDRUtil, one SoapySDRUtil and one CubicSDR,...

gdb trace for the server : 
```
(gdb) file /usr/local/bin/SoapySDRServer
Reading symbols from /usr/local/bin/SoapySDRServer...done.
(gdb) run --bind
Starting program: /usr/local/bin/SoapySDRServer --bind
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.6.0-gc09b2f10
Server UUID: 3803b9fa-e23d-1658-8567-22a9007f0101
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server... 
[New Thread 0x7ffff7a01700 (LWP 8874)]
Connecting to DNS-SD daemon... 
[WARNING] SoapyRemote compiled without DNS-SD support!
Press Ctrl+C to stop the server
SoapyServerListener::accept([::ffff:127.0.0.1]:42690)
[New Thread 0x7ffff7200700 (LWP 8877)]
SoapyServerListener::accept([::ffff:127.0.0.1]:42692)
[New Thread 0x7ffff69ff700 (LWP 8878)]
[Thread 0x7ffff7200700 (LWP 8877) exited]
[Thread 0x7ffff69ff700 (LWP 8878) exited]
SoapyServerListener::accept([::ffff:127.0.0.1]:42694)
[New Thread 0x7ffff5e3b700 (LWP 8881)]
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:127.0.0.1]:42696)
[New Thread 0x7ffff69ff700 (LWP 8882)]
[New Thread 0x7ffff7200700 (LWP 8887)]
[New Thread 0x7ffff553d700 (LWP 8888)]
[New Thread 0x7ffff4d3c700 (LWP 8889)]
[New Thread 0x7fffdffff700 (LWP 8890)]
[New Thread 0x7fffdf7fe700 (LWP 8891)]
[Thread 0x7fffdf7fe700 (LWP 8891) exited]
SoapyServerListener::accept([::ffff:127.0.0.1]:42702)
[New Thread 0x7fffdeffd700 (LWP 8893)]
SoapyServerListener::accept([::ffff:127.0.0.1]:42704)
[New Thread 0x7fffde7fc700 (LWP 8894)]
[Thread 0x7fffdeffd700 (LWP 8893) exited]
[Thread 0x7fffde7fc700 (LWP 8894) exited]
SoapyServerListener::accept([::ffff:127.0.0.1]:42706)
[New Thread 0x7fffddffb700 (LWP 8896)]
SoapyServerListener::close()
SoapyServerListener::close()
[... 19 lines cut ...]
[Thread 0x7fffc2ffd700 (LWP 8910) exited]

Thread 10 "SoapySDRServer" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffdffff700 (LWP 8890)]
__memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:384
384     ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S: Aucun fichier ou dossier de ce type.
(gdb) bt
#0  __memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:384
#1  0x00007ffff609627f in SoapySDRPlay::readStream (this=0x7fffec002970, stream=0x7fffec001c10, buffs=0x7fffd4000b20, numElems=1006, 
    flags=@0x7fffdfffe99c: 0, timeNs=@0x7fffdfffe990: 0, timeoutUs=100000)
    at ~/compile/SoapySDRPlay3/Streaming.cpp:418
#2  0x00005555555bb814 in ServerStreamData::sendEndpointWork (this=0x7fffec002168)
    at ~/compile/SoapyRemote/server/ServerStreamData.cpp:185
#3  0x00005555555bc198 in std::__invoke_impl<void, void (ServerStreamData::*)(), ServerStreamData*> (
    __f=@0x7fffec0055a0: (void (ServerStreamData::*)(ServerStreamData * const)) 0x5555555bb554 <ServerStreamData::sendEndpointWork()>, __t=@0x7fffec005598: 0x7fffec002168) at /usr/include/c++/8/bits/invoke.h:73
#4  0x00005555555bbb3b in std::__invoke<void (ServerStreamData::*)(), ServerStreamData*> (
    __fn=@0x7fffec0055a0: (void (ServerStreamData::*)(ServerStreamData * const)) 0x5555555bb554 <ServerStreamData::sendEndpointWork()>, __args#0=@0x7fffec005598: 0x7fffec002168) at /usr/include/c++/8/bits/invoke.h:95
#5  0x00005555555bcf8b in std::thread::_Invoker<std::tuple<void (ServerStreamData::*)(), ServerStreamData*> >::_M_invoke<0ul, 1ul> (
    this=0x7fffec005598) at /usr/include/c++/8/thread:244
#6  0x00005555555bcf46 in std::thread::_Invoker<std::tuple<void (ServerStreamData::*)(), ServerStreamData*> >::operator() (
    this=0x7fffec005598) at /usr/include/c++/8/thread:253
#7  0x00005555555bcf2a in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (ServerStreamData::*)(), ServerStreamData*> > >::_M_run (this=0x7fffec005590) at /usr/include/c++/8/thread:196
#8  0x00007ffff7e3bb2f in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007ffff7f0bfa3 in start_thread (arg=<optimized out>) at pthread_create.c:486
#10 0x00007ffff7b194cf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95```

nmaster2042, 2021-01-10:
I make some test, for now only single tuner mode, soapySDRRemote.

Same test as you @fventuri, no issues.

With Cubic, start firts one ST mode, Tuner 1, center freq = 7500 khz, ok
Start second cubic, I see RSP duo ST avaliable, select it same Tuner, frequency, sample rate, ok.
My 2 instances are running, if I stop the stream of one of the two, they crash togather,

On my terminal of instance 1 I see 

`
[ERROR] SoapyLogAcceptor::handlerLoop(tcp://192.168.0.65:55132) FAIL: SoapyRPCUnpacker::recv(header) FAIL: 
terminate called after throwing an instance of 'std::runtime_error'
  what():  SoapyRPCUnpacker::recv(header) FAIL: 
Abandon (core dumped)
`
And on the second instance terminal

`
[ERROR] SoapyLogAcceptor::handlerLoop(tcp://192.168.0.65:55132) FAIL: SoapyRPCUnpacker::recv(header) FAIL: 
Terminating SDR thread..
ERROR: thread '9SDRThread' has not terminated in time ! (> 3000 ms)
`

Finally my SoapySDRServer crash too.

nmaster2042, 2021-01-10:
Humm I tried the same approach with as airspyhf+, opening 2 instances of cubicsdr and stream same frequency.
It works but issues also when I stop one of the two.

So it's possible issues are not only coming from SoapySDRPlay3.

nmaster2042, 2021-01-12:
@fventuri viewing @pausrn's tests, I was wondering if it's ok to start different streams with different sample rates ?
I thought it had to be the same ?

nmaster2042, 2021-01-12:
@pausrn I tried this 2 streams on my system:

`SoapySDRUtil --args="driver=remote,remote=192.168.0.65,remote:driver=sdrplay,remote:prot=tcp" --rate=6e6 --direction=RX`

and

`SoapySDRUtil --args="driver=remote,remote=192.168.0.65,remote:driver=sdrplay,remote:prot=tcp" --rate=4e6 --direction=RX`

No crash here.

If I change the second stream with a sample rate >= 5 Mbps it crash

fventuri, 2021-01-13:
@nmaster2042 , @pausrn - thanks for all your comments and tests.

In the last couple of days I have been running some tests with a very simple RSP2 device, and I was able to reproduce the crashes using any value of the sample rate > 2MHz.
I see that SoapySDRServer calls `setSampleRate()` every time a new client stream is started, so if I run `SoapySDRUtil` with the argument `--rate=2.1e6`, and, while it is running, I start another instance of `SoapySDRUtil` in another windows with the argument `--rate=2.2e6`, then the second call to `setSampleRate()` with 2.2e6 overrides the first call, and I suspect that even the first stream gets a new sample rate of 2.2e6 (unless SoapySDRServer has some decimation/interpolation code inside; I haven't checked that).

As per the crashes, they seem to be due to incorrect handling of buffer overflows in the SoapySDRPlay3 driver; I am still trying to understand how that is supposed to work in first place; after I have that figured out, I should able to find the problem and fix it.

Franco

fventuri, 2021-01-15:
@nmaster2042 @pausrn - last night I figured out that the problem (at least here) is that SoapySDRServer is using multiple threads to read samples from the RSP using the `readStream()` SoapySDR function.
I also did some debugging on `readStream()` and it looks like the way it is now it has problems with multithreading (for instance in the block where it calls the function `acquireReadBuffer()`, and probably in other parts).

Based on those findings from last night, this morning I added (for now) another mutex that prevents more than one thread from running `readStream()` at the same time; I ran a couple of quick tests, and this time `SoapySDRServer` did not crash for me; I just pushed the code with the fix to the usual branch `support_for_serial_keyword` for you to run your tests.

This mutex is just a temporary band-aid for now until I can take a better look at what parts of `readStream()` do not support multithreading; once I have that figured out, I should be able to put the proper multithreaded protection in the right places, but for now I'd like to find out if it addresses the problem with the crashes for you too.

Franco

@nmaster2042 - Thanks for the screenshots with gqrx; I received the links this morning, and I'll have a  look at them after I am done with this problem with multithreading in `readStream()`

nmaster2042, 2021-01-18:
@fventuri, I mkde new tests with your last commit.

Now I can stream 2 instances with SoapyUtil:

`SoapySDRUtil --args="driver=remote,remote=192.168.0.65,remote:driver=sdrplay,remote:prot=tcp" --rate=6e6 --direction=RX`

And

`SoapySDRUtil --args="driver=remote,remote=192.168.0.65,remote:driver=sdrplay,remote:prot=tcp" --rate=5e6 --direction=RX`

No more crash

So I tried with 2 instances of cubicSDR.

I start a first instance, Tuner 1 50 ohms, 1M SR and a 2nd one with sale device, tuner and SR, all ok.

But when I stop the first instances, both Cubic instances crashes:

Instance 1:

```
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
readLoop() ended.
[ERROR] SoapyLogAcceptor::handlerLoop(tcp://192.168.0.65:55132) FAIL: SoapyRPCUnpacker::recv(header) FAIL: 
terminate called after throwing an instance of 'std::runtime_error'
  what():  SoapyRPCUnpacker::recv(header) FAIL: 
Abandon (core dumped)
```

Instance 2:

```
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code SOAPY_SDR_STREAM_ERRORSDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
[... cut ...]
```
I have to kill process.

Then instance 2 / slave blocks too: It continues until I stop stream but I can't restart it, same error as master:

```
[ERROR] error in activateStream() - Init() failed: sdrplay_api_AlreadyInitialised
SDRThread : Device Stream set to MTU: 357
starting readLoop()
ERROR: thread '9SDRThread' has not terminated in time ! (> 2000 ms)
```

We don't have same behavour with command line and Gui app, so I don't knwo if the issue is related to the soapySDRPlay3 driver or cubic.

fventuri, 2021-01-18:
@nmaster2042 - just to make sure I understand and can reproduce your problem correctly, are you using SoapyRemote to connect to your RSPduo from CubicSDR (i.e. selecting a remote SoapySDR device), or are connecting to the RSPduo directly from CubicSDR (i.e. using the device under 'Local')?

Franco

nmaster2042, 2021-01-18:
@fventuri - Initially I used my RSPDUO locally with Gqrx.

15 days ago, I moved the device to a remote server, as you asked tests with SoapySDRServer, specially for Rspduo.

So actually I make all my tests with SoapySDRServer.

There is issues with Gqrx and SorpySDRServer, that's why I switched to Cubic when starting tests with remote server.

I hope it's more understandable for you.

fventuri, 2021-01-18:
@nmaster2042 - thanks for the thorough explanation of your environment; here instead I am using just one computer and, when I want to test the SoapySDRPlay3 driver using SoapyRemote, I just comment a couple of lines in CubicSDR where it enumerates the local SDRs (https://github.com/cjcliffe/CubicSDR/blob/master/src/sdr/SDREnumerator.cpp#L189-L190), so it won't see them as local, and I can just use SoapyRemote with an IP address of 127.0.0.1 (i.e. localhost).

Anyway, I think I figured out the problem with the two instances of CubicSDR using SoapyRemote to see the RSPduo in Single Tuner mode (it's due to the fact that it is calling `deactivateStream()` multiple times, so I had to move the reference count of the active streams to `closeStream()` instead).

I just pushed this fix to the usual `support_for_serial_keyword` branch; please give it a try and let me know how it goes.

Franco

nmaster2042, 2021-01-19:
@fventuri, it seems you got it with the last commit.

I remade the 2 x RSPDuo single Tuner mode.

First I started an instance of Cubic, selected remote RSPDuo Single Tnuer mode, Tuner 1, SR 2m and stream.
Then I started a 2nd instance of Cubic, ans made same choice as for instance 1, then stream.

I got 2 Stream from same tuner, all ok.

Then, tried to stop stream of instance 1, stopped successfully, instance 2 continues streaming.
I restarted streaming on instance 1, restarts successfully, instance 2 continues streaming.

With my 2 instances streaming, stopped streaming of instance 2, stopped successfully, instance 1 continues streaming.
I restarted streaming on instance 2, restarts successfully, instance 1 continues streaming.

I only noticed error message on the cubic instance 1 console when first started and enumerating devices, see below:

```
  driver = remote
  label = SDRplay Dev1 RSPduo 1809029C32 - Dual Tuner
  mode = DT
  remote = tcp://192.168.0.65:55132
  remote:driver = sdrplay
  serial = 1809029C32
Make device 3
[INFO] devIdx: 0
[INFO] hwVer: 3
[INFO] rspDuoMode: 2
[INFO] tuner: 3
[INFO] rspDuoSampleFreq: 6000000.000000
  remote:version=0.6.0-gc09b2f10
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=3

  driver = remote
  label = SDRplay Dev2 RSPduo 1809029C32 - Master
  mode = MA
  remote = tcp://192.168.0.65:55132
  remote:driver = sdrplay
  serial = 1809029C32
Make device 4
[INFO] devIdx: 0
[INFO] hwVer: 3
[INFO] rspDuoMode: 4
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 6000000.000000
  remote:version=0.6.0-gc09b2f10
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=3

  driver = remote
  label = SDRplay Dev3 RSPduo 1809029C32 - Master (RSPduo sample rate=8Mhz)
  mode = MA8
  remote = tcp://192.168.0.65:55132
  remote:driver = sdrplay
  serial = 1809029C32
Make device 5
[INFO] devIdx: 0
[INFO] hwVer: 3
[INFO] rspDuoMode: 4
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 8000000.000000
  remote:version=0.6.0-gc09b2f10
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=3
[ERROR] ReleaseDevice Error: sdrplay_api_ServiceNotResponding
[ERROR] SoapyLogAcceptor::handlerLoop(tcp://192.168.0.65:55132) FAIL: SoapyRPCUnpacker::recv(header) FAIL: 
[ERROR] ~SoapyRemoteDevice() FAIL: SoapyRPCUnpacker::recv(header) FAIL: 
[ERROR] SoapyLogAcceptor::connect(tcp://192.168.0.65:55132) FAIL: connect(tcp://192.168.0.65:55132) [111: Connexion refusée]
[ERROR] SoapyLogAcceptor::shutdown(tcp://192.168.0.65:55132) FAIL: SoapyRPCPacker::send() FAIL: send() [32: Relais brisé (pipe)]

Reporting enumeration complete.
SDR enumerator done.
[INFO] devIdx: 0
[INFO] hwVer: 3
[INFO] rspDuoMode: 1
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
SDR thread starting.
device init()
```

In your last reply, you only mentioned the Single Tuner use case  you corrected, so I didn't test the master / slave.

Let me know if/when I can test it.

fventuri, 2021-01-19:
@nmaster2042 - first of all, thanks for running all those tests!

I just ran a couple of very simple tests (starts and stops) here going through SoapyRemote with the RSPduo both in Master and Slave mode, and SoapySDRServer did not crash on my computer (of course the Master instance needs to be running for the Slave instance to be working, as per SDRplay documentation), so at this point please give it a try there, and let me know how it goes (if it works, then we can start looking at the problems with gqrx).

As per those `FAIL: SoapyRPCUnpacker::recv(header)` messages, I see them here too; I am not really sure what they mean and where they come from, but they do not seem to be showstoppers (at least here), because even after those messages I can start streaming again without any problems.

Franco

nmaster2042, 2021-01-20:
@fventuri - I made the Master/Slave test with Cubic and SoapySDRRemote.

I started a first instance of Cubic, selected RSPDuo master, tuner 1 50 ohms, sr 2M, streams ok.
Then I started a 2nd instance of Cubic, selected RSPDuo slave, rs 2M, streams ok.

Instance 2 automatically used tuner 2.

Test 1: slave instance, stop stream - OK, instance 1 (master) continues streaming.
Test 2 : slave instance, start stream again - OK, instance 1 (master) continues streaming.
Test 3 : master instance, stop stream, OK, instance 2 (slave) continues streaming.
Test 4 : master instance, start stream again - FAIL, Cubic stays frozen, instance 2 (slave) continues streaming.

At this point, in the instance 1 (master) console, I get an error:

```
[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CF32, scaleFactor=32767, mtu=1500, window=44040192)
[INFO] Client side stream bound to 192.168.0.3:37832
[INFO] Client side status bound to 192.168.0.3:50763
[INFO] Using format CS16.
[INFO] Server side stream bound to [::ffff:192.168.0.65]:47223
[INFO] Client side stream connected to 192.168.0.65:47223
[INFO] Configured receiver endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=43008 KiB
[INFO] Server side stream connected to [::ffff:192.168.0.3]:37832
[INFO] Server side status connected to [::ffff:192.168.0.3]:50763
[WARNING] StreamEndpoint resize socket buffer: set 43008 KiB, got 208 KiB
[INFO] Configured sender endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=208 KiB
[ERROR] error in activateStream() - Init() failed: sdrplay_api_AlreadyInitialised

```
I have to kill instance 1, it makes instance 2 to freeze too, and on the instance 2 (slave) console I get this:

```
[INFO] Using format CS16.
[INFO] Server side stream bound to [::ffff:192.168.0.65]:47223
[INFO] Server side stream connected to [::ffff:192.168.0.3]:37832
[INFO] Server side status connected to [::ffff:192.168.0.3]:50763
[WARNING] StreamEndpoint resize socket buffer: set 43008 KiB, got 208 KiB
[INFO] Configured sender endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=208 KiB
[ERROR] error in activateStream() - Init() failed: sdrplay_api_AlreadyInitialised
Terminating SDR thread..
ERROR: thread '9SDRThread' has not terminated in time ! (> 3000 ms)
```
Finally I have to kill instance 2 too.

nmaster2042, 2021-01-20:
I discovered an other issue: When I start Cubic, I get the device selection window.
I click on one of RSPDuo device, it's ok but if I click on another RSPDuo device it makes Cubic crash with this error in the console:

```
  driver = airspyhf
  label = AirSpy HF+ [dd52eb5df538399e]
  serial = dd52eb5df538399e
Make device 0
  serial=dd52eb5df538399e

  driver = remote
  label = AirSpy HF+ [dd5282aa314c3f36]
  remote = tcp://192.168.0.65:55132
  remote:driver = airspyhf
  serial = dd5282aa314c3f36
Make device 1
  remote:version=0.6.0-gc09b2f10
  serial=dd5282aa314c3f36

  driver = remote
  label = SDRplay Dev0 RSPduo 1809029C32 - Single Tuner
  mode = ST
  remote = tcp://192.168.0.65:55132
  remote:driver = sdrplay
  serial = 1809029C32
Make device 2
[INFO] devIdx: 0
[INFO] hwVer: 3
[INFO] rspDuoMode: 1
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
  remote:version=0.6.0-gc09b2f10
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=3

  driver = remote
  label = SDRplay Dev1 RSPduo 1809029C32 - Dual Tuner
  mode = DT
  remote = tcp://192.168.0.65:55132
  remote:driver = sdrplay
  serial = 1809029C32
Make device 3
[INFO] devIdx: 0
[INFO] hwVer: 3
[INFO] rspDuoMode: 2
[INFO] tuner: 3
[INFO] rspDuoSampleFreq: 6000000.000000
  remote:version=0.6.0-gc09b2f10
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=3

  driver = remote
  label = SDRplay Dev2 RSPduo 1809029C32 - Master
  mode = MA
  remote = tcp://192.168.0.65:55132
  remote:driver = sdrplay
  serial = 1809029C32
Make device 4
[INFO] devIdx: 0
[INFO] hwVer: 3
[INFO] rspDuoMode: 4
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 6000000.000000
  remote:version=0.6.0-gc09b2f10
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=3

  driver = remote
  label = SDRplay Dev3 RSPduo 1809029C32 - Master (RSPduo sample rate=8Mhz)
  mode = MA8
  remote = tcp://192.168.0.65:55132
  remote:driver = sdrplay
  serial = 1809029C32
Make device 5
[INFO] devIdx: 0
[INFO] hwVer: 3
[INFO] rspDuoMode: 4
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 8000000.000000
  remote:version=0.6.0-gc09b2f10
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=3

Reporting enumeration complete.
SDR enumerator done.
[INFO] devIdx: 0
[INFO] hwVer: 3
[INFO] rspDuoMode: 1
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
Terminating SDR thread..
ERROR: thread '9SDRThread' has not terminated in time ! (> 3000 ms)
```

fventuri, 2021-01-22:
@nmaster2042 
The first of the problems you mentioned yesterday is due to the fact that, according to SDRplay documentation, the RSPduo slave instance depends on the RSPduo master instance to run; because of that I suspect that stopping the RSPduo master instance while the RSPduo slave is still attached might cause problems.

To prevent this scenario, I just added a few lines of code to the SoapySDRPlay3 driver that should output a WARNING message and ignore the request to close the instance, when the user tries to stop the RSPduo master while a slave is still attached.

I ran a few quick tests stopping and starting master and slave in CubicSDR using SoapyRemote with this new code changes, similar to what you did in Tests 1-4, and I didn't have any crash of freeze here, so I just pushed the new code to the usual `support_for_serial_keyword` branch, for you (and anyone who wants to run tests) to try it too.

As per the second problem you posted (the one about clicking on two incompatible modes for the same RSPduo), I didn't realize that CubicSDR allowed you to do that (it's probably because CubicSDR thinks those pseudo-devices are completely independent, while in reality they are not). I'll look into it tomorrow night and I'll try to figure out if there is an easy way to prevent that.

Franco

nmaster2042, 2021-01-22:
@fventuri  I made new tests with your last commit and it's fine now.

If I stop master, it actually stops, and see warnings you added in the console of master instance.

```
[WARNING] error in closeStream() - trying to close the RSPduo master while the slave is still attached
[WARNING] error in ~SoapySDRPlay() - trying to run the SoapySDRPlay destructor for the RSPduo master while the slave is still attached

```
But, now when I stop and close my 2 instances of master slave, when restart cubic, I only get RSPDuo master and slave devices, like when I start a 2nd instance and RSPDuo master is already in use.

It seems something is not closed correctly.

fventuri, 2021-01-28:
@nmaster2042
I tried a few different approaches during the weekend and in the last couple of days, but I was not able to find a clean way to handle the case when an RSPduo master is terminated before the slave; the code I wrote checks for the RSPduo streaming event `sdrplay_api_MasterDllDisappeared` so the slave would know when the master is terminated, but that event is never triggered.

My most recent idea is to have the RSPduo master to wait for the slave to end streaming before terminating; I implemented a while loop that, when the client application attempts to terminate the RSPduo master, checks every few seconds that the is no slave device still attached, and displays the warning message `Please close RSPduo slave device first. Trying again in 10 seconds` until the slave is gone. This approach seems to work correctly with SoapyRemote and SoapySDRUtil:
```
# SoapyRemote
SoapySDRServer --bind

# Master client
SoapySDRUtil --args="driver=remote,remote=127.0.0.1,remote:driver=sdrplay,remote:mode=MA,remote:prot=tcp" --rate=2.048e6 --direction=RX

# Slave client
SoapySDRUtil --args="driver=remote,remote=127.0.0.1,remote:driver=sdrplay,remote:mode=SL,remote:prot=tcp" --rate=2.048e6 --direction=RX
```

However with CubicSDR, it looks like the application running in master mode still shuts down after a few seconds (i.e. before I have the time to stop the application running in slave mode), and this behavior leaves the SoapyRemote server in an unstable state.

I pushed these latest changes to the usual `support_for_serial_keyword` branch, so you can give it a try, if you like.

Franco

SDRplay, 2021-01-28:
@fventuri we capture this in SDRuno in two ways...

first, in the master application, if it's tried to be closed with the slave still attached then we put up a message saying that this application cannot close before the slave. You can use the following code in the event callback to that effect...

switch(eventId)
{
   case sdrplay_api_RspDuoModeChange:
      if(params->rspDuoModeParams.modeChangeType == sdrplay_api_SlaveAttached)
      {
         slaveAttached = true;
      }
      if(params->rspDuoModeParams.modeChangeType == sdrplay_api_SlaveDetached)
      {
         slaveAttached = false;
      }
      break;
}

Then on the slave side, if the master has disappeared, we use the following code in the same event callback switch case...

else if(params->rspDuoModeParams.modeChangeType == sdrplay_api_MasterDllDisappeared)
{
   // display message saying that the master no longer exists and the slave will be forced to shut down.
}

now I know it's slightly different because this is a library that is used within another application, but hopefully there is a mechanism by which the hardware library can notify the higher level application that there is an error and the application needs to exit?

Hope that helps

Andy

fventuri, 2021-01-28:
@SDRplay
Thanks for the very useful comments Andy; as they say "great minds thinks alike".

- the code you show for the slave side (i.e. the one with `sdrplay_api_MasterDllDisappeared`) is very similar (if not identical) to the code in the SoapySDRPlay driver (https://github.com/pothosware/SoapySDRPlay3/blob/master/Streaming.cpp#L179); however in the scenario we are testing (see below) the only two RspDuoModeChange events I see are `sdrplay_api_SlaveInitialised` and `sdrplay_api_SlaveUninitialised` (both on the master side; the second one only when I stop the slave before the master, of course)
- the idea of using the `slaveAttached` boolean variable to keep the status is what I tried in this commit about a week ago (https://github.com/pothosware/SoapySDRPlay3/commit/40a3c8606aeb5f0e9d94011a552081d66fe09fc6#diff-cabbe0d289c7cccedc8350e2da8279a912ec148c512cbf0e24ddbfa6f488e658); however our tests showed that it still caused problems (crashes and freezes) in our test scenarios
- another very similar approach that avoids the use of a variable to keep the status of the slave in the master, is to check the return value of the `sdrplay_api_Uninit()` API call; if it returns the code `sdrplay_api_StopPending`, it means that the slave is still attached; this is what I implemented last night in the `support_for_serial_keyword` branch (see here: https://github.com/pothosware/SoapySDRPlay3/blob/support_for_serial_keyword/Streaming.cpp#L283-L293); however some clients like CubicSDR wait only a few seconds before shutting down anyway, thus never telling the SoapyRemote server that they are done, and therefore `sdrplay_api_ReleaseDevice` never gets called for the master
- just to make sure we are all on the same page, the use case scenarios @nmaster2042 and I are testing here are using SoapySDRServer (SoapyRemote) as the remote server between the client and the RSPduo, and using two different types of streaming clients:
  - the command line client `SoapySDRUtil` (which can do streaming) - things seem to work here
  - the GUI application CubicSDR - here is where we have problems
- the use case scenario we are testing is the user starting a master, starting a slave, and then attempting to stop the master before the slave

Franco

nmaster2042, 2021-01-31:
@fventuri  I ran new tests with your last commit.

Now things are like this:

Test 1: slave instance, stop stream - OK, instance 1 (master) continues streaming.
Test 2 : slave instance, start stream again - OK, instance 1 (master) continues streaming.

No change for these 1 use cases. Let's continue

Test 3 : master instance, stop stream, OK, instance 2 (slave) continues streaming.
On terminal of the master instance we have this warning:

`[WARNING] Please close RSPduo slave device first. Trying again in 10 seconds`

Test 4 : master instance, start stream again, now it actually restarts fine.

And to do things right, I must stop slave first.

I think it's the vest that can be done for this particular case.

fventuri, 2021-02-02:
I agree with you @nmaster2042 - I need to look at a different issue with the GNU Radio module gr-sdrplay3 these days; after I am done with that, I plan to run some more tests with just CubicSDR and gqrx (without SoapyRemote) to make sure there are no more cases where the SoapySDRPlay driver crashes.

If everything looks good, then I think this branch can finally be merged into master.

Franco

nmaster2042, 2021-02-02:
@fventuri OK. Let me know if you need help for tests, specially with Gqrx. With SoapyRemote I experienced waterfall freezes, when changing demodulator or if I stop stream and start again.

fventuri, 2021-02-09:
I resumed working on this branch tonight (this way I hope I'll be able to merge it soon into master), and I found another issue when switching RSPduo mode inside CubicSDR (both directly and with SoapySDRRemote in between).

These are the steps to reproduce the problem:
- start CubicSDR and select say Single Tuner mode; start streaming
- after a while stop streaming; navigate to the CubicSDR menu 'SDR Devices', and choose say 'Master' instead
- CubicSDR crashes due to an exception thrown by the SoapySDRPlay driver

The reason is that the RSPduo needs to be released (with the API function `sdrplay_api_ReleaseDevice`) before it can be selected again with a different RSPduo mode. Right now the only place where the SoapySDRPlay driver calls `sdrplay_api_ReleaseDevice` is in the SoapySDRPlay destructor (via the `releaseDevice()` function), but that destructor never gets called by CubicSDR (or by SoapySDRServer); the only functions that are called are `deactivateStream()` and `closeStream()`.

In order to make things work as a normal user would expect, one option could be to check in the SoapySDRPlay constructor if that RSPduo has already been selected from a previous call to the constructor; if it has, the code should first call the releaseDevice() on that instance and then finally select it again with the new chosen RSPduo mode. The one problem I see with this workaround is that the old instance of the SoapySDRPlay device is still around (since it has not been destroyed), and therefore the client application may think it is still available and expect for instance to be able to stream with it (by calling `activateStream()`) .

Not really sure if this is a good idea, but it's getting late here, and I need to think about it a bit longer.

Any ideas or suggestion on how to make this whole thing more robust is most welcome!

Franco

SDRplay, 2021-02-09:
@fventuri have you tried sdrplay_api_SwapRspDuoMode?

Best regards,

Andy

fventuri, 2021-02-09:
Thanks for the tip @SDRplay !

I was not aware about that API function; I just looked into the API 3.06 documentation (https://www.sdrplay.com/docs/SDRplay_API_Specification_v3.06.pdf) and it is not listed there; however I also checked the API include files and I do see it in `sdrplay_api.h`:

```
    _SDRPLAY_DLL_QUALIFIER sdrplay_api_ErrT        sdrplay_api_SwapRspDuoMode(sdrplay_api_DeviceT *currDevice, sdrplay_api_DeviceParamsT **deviceParams,
                                                                              sdrplay_api_RspDuoModeT rspDuoMode, double sampleRate, sdrplay_api_TunerSelectT tuner,
                                                                              sdrplay_api_Bw_MHzT bwType, sdrplay_api_If_kHzT ifType, sdrplay_api_RspDuo_AmPortSelectT tuner1AmPortSel);
```

Tonight after work I'll give it a try; given the parameter names, I think I can figure out how to make it work, but I was wondering if the PDF I mentioned above is not the latest version of the API specification.

Thanks again,
Franco

SDRplay, 2021-02-09:
It is a little used function in the API and it wasn't there when the documentation was first put together. You are right, the API document needs updating to include this function. It's on my ToDo list. If you have a problem using it, let me know. It's not a function that I use in SDRuno. It was implemented after I had my own way of doing it and I've never got round to using it.

fventuri, 2021-02-11:
@SDRplay

My first attempt tonight at calling `sdrplay_api_SwapRspDuoMode` didn't go so well.

I created a new method called `swapRspDuoMode`, so I can segregate all the new code to invoke `sdrplay_api_SwapRspDuoMode` inside that method and avoid making the SoapySDRPlay constructor code even more spaghetti-like.

I then ran the usual three step test with CubicSDR:

1. start CubicSDR and select Single Tuner mode; start streaming
2. after a while stop streaming (File -> Start/Stop Device)
3. navigate to the CubicSDR menu 'SDR Devices', and select 'Master' instead

I do see that the new method is called in step 3 as expected, but when the method calls `sdrplay_api_SwapRspDuoMode`, it crashes; I made it print the arguments right before the API call and they seem 'reasonable' to me (but it is getting late here, so I might have missed something obvious):

```
[INFO] swapRspDuoMode() - before SDRplay API call - &device=0x35372d0, &deviceParams=0x3537330 rspDuoMode=4 rspDuoSampleFreq=6000000.000000 tuner=1 bwType=1536 ifType=1620 tuner1AmPortSel=0
```

I just pushed this new code to a new branch called 'swap_rspduo_mode' (https://github.com/pothosware/SoapySDRPlay3/tree/swap_rspduo_mode); please take a look and tell me what you think; the code compiles without any errors or warnings, so you should be able to run it there too, if you would like to reproduce the problem.

Thanks for all your help,
Franco

SDRplay, 2021-02-11:
@fventuri when you stop the device, do you Uninit? If so, that could be the problem. The function is designed to be run whilst the device is still in the Init state.

If you are still having problems, please send me the API debug output and I'll take a closer look.

Best regards,

Andy

fventuri, 2021-02-11:
@SDRplay 
You re right, Andy; when CubicSDR stops streaming, it makes two SoapySDR API calls: first `deactivateStream()`, and then `closeStream()`, where I do call `sdrplay_api_Uninit()`, so that would explain the crash.

If I understand correctly then, the function `sdrplay_api_SwapRspDuoMode()` is designed for the scenario when the user wants to switch RSPduo mode while they are streaming (which honestly is something I never thought one would do based on the abstraction of pseudo-devices that the SoapySDRPlay driver presents to the client application), while in my test above I was trying to go through the steps of someone simply wanting to 'switch' the SDR device they are using in CubicSDR.

At this point I think I need to come up with a comprehensive list of use-case scenarios that the user might reasonably expect to try with say CubicSDR, gqrx, and SoapyRemote, figure out which SoapySDR calls are invoked in each case, and finally try to map those calls to the SDRplay APIs functions like SelectDevice, ReleaseDevice, Init, Uninit, SwapRspDuoMode, etc

Franco

SDRplay, 2021-02-11:
@fventuri it really comes down to how you see the devices and it's slightly more complicated in SoapySDR framework case because of not having the control over the end application.

You could take the view that you can't switch these modes without stopping the application and restarting it. Or if you do want to try to use the switch modes function, you will need to do it whilst the device is still initialised.

Andy

nmaster2042, 2021-02-15:
@fventuri I built your last commit of swap_rspduo_mode and tried with my RSP Duo over network SoapySDRServer.

I get same result as you with cubicsdr.

When I run some time on Single Tuner mode, Stop and select master mode, cubic sdr crashes with this error.

```
[INFO] swapRspDuoMode() - swapRspDuo=0xffff64012570 - args=agc_setpoint=-30, biasT_ctrl=true, dabnotch_ctrl=false, driver=sdrplay, extref_ctrl=false, iqcorr_ctrl=true, label=SDRplay Dev2 RSPduo 1809029C32 - Master, mode=MA, remote=tcp://192.168.0.65:55132, rfgain_sel=4, rfnotch_ctrl=false, serial=1809029C32, soapy_remote_no_deeper=
[INFO] rspDuoMode: 4
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 6000000.000000
[INFO] swapRspDuoMode() - before SDRplay API call - &device=0xffff4c003200, &deviceParams=0xffff4c003260 rspDuoMode=4 rspDuoSampleFreq=6000000.000000 tuner=1 bwType=1536 ifType=1620 tuner1AmPortSel=0
[ERROR] SoapyLogAcceptor::handlerLoop(tcp://192.168.0.65:55132) FAIL: SoapyRPCUnpacker::recv(header) FAIL: 
Terminating SDR thread..
Terminating SDR post-processing thread..
Terminating All Demodulators..
Terminating Visual Processor threads..
Application termination complete.
Unhandled unknown exception; terminating the application.
```

fventuri, 2021-02-15:
Thanks for trying it @nmaster2042 

Over the weekend I came up with a perhaps cleaner solution that I started coding, and I should be able to debug in the next couple of nights after work.
Once it works reasonably here, I'll push it to the repository so you can give it a try, and see if it works reasonably OK there too.

Franco

fventuri, 2021-03-25:
@nmaster2042 
I rewrote a good portion of the RSP device selection code, and in my initial tests here these changes seem to address the problems with crashes and freezes in `CubicSDR` (and hopefully in `gqrx` and with `SoapySDRRemote`), when switching between different RSPduo modes (single tuner to master, and viceversa). These changes however do not prevent CubicSDR to crash when it is streaming from the RSPduo in slave mode, and the user suddenly stops the application that is running in master node (I am not even sure if there is a good way to deal with this scenario).

Anyway, I just pushed all these changes to the usual `support_for_serial_keyword` branch for you to try out to see if things are better now.

If it looks OK, then we can go ahead and finally merge this branch into `master` (long due!)

I apologize for making you wait for more than a month for it.

Franco

vsonnier, 2021-03-25:
Hi @fventuri, thanks for your hard work ! 

I give the branch 'support_for_serial_keyword' a try with my RSP2 under Windows and CubicSDR, using the recompiled SoapySDR `master` as well.

The fist run happens correctly, howver when I re-launch CubicSDR, I got the following errors:

Second run:
```
SDR enumerator starting.
SoapySDR init..
        API Version: v0.8.0
        ABI Version: v0.8
        Install root: D:\Projets\GIT_Repositories\CubicSDR\CubicSDR_win64
        Loading modules...
Loading bundled SoapySDR module D:\Projets\GIT_Repositories\CubicSDR\CubicSDR_win64\x64/modules//sdrPlaySupport.dll..
        Available factories...sdrplay
  driver = sdrplay
  label = SDRplay Dev0 RSP2 1702007700
  serial = 1702007700
Make device 0
[INFO] devIdx: 0
[INFO] hwVer: 2
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0,000000
  sdrplay_api_api_version=3,070000
  sdrplay_api_hw_version=2
Initializing WGL swap control extensions..
Loaded font 'Bitstream Vera Sans Mono' from ...
Reporting enumeration complete.
SDR enumerator done.
[INFO] devIdx: 0
[INFO] hwVer: 2
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0,000000
SDR thread starting.
device init()
[INFO] Using format CF32.
[ERROR] error in activateStream() - Init() failed: sdrplay_api_Fail
SDRThread : Device Stream set to MTU: 65536
starting readLoop()
...
```  
No amount of plug-unplug of the device, or restarting the service cures it, I have to reboot the machine to get the RSP2 back.

vsonnier, 2021-03-25:
Hold on, my RSP2 may have chosen to die today... I'll investigate on my side first.

fventuri, 2021-03-25:
@vsonnier 
This morning I plugged in the RSP2 (I ran last night's initial tests only on the RSPduo), and I was able to stream from it, stop and start it several times without any crashes or freezes.
I should have mentioned that I am using Linux here for all my tests, while I see you are running Windows, so it is very possible that I missed something (and hopefully it is not a problem with the hardware).

Let me know what you find out,
Franco

vsonnier, 2021-03-25:
> @vsonnier
> This morning I plugged in the RSP2 (I ran last night's initial tests only on the RSPduo), and I was able to stream from it, stop and start it several times without any crashes or freezes. [...]

Thanks Franco to have tried on your side. Luckily, I saw quickly after my last message that SDRUno worked perfectly... After a full resinstall of SDRUno 1.40.2, Cubic still refused to work.... 

Until I finally cleared the CubicSDR `config.xml` manually. Probably since last time I updated the module, the `device  - id` string would have changed enough to confuse the app...

Both Win32 and Win64 versions of CubicSDR (and the corresponding modules) seems to work.

Sorry Franco for the unecessary scare, thanks again for your efforts.

fventuri, 2021-03-26:
@vsonnier 
Thanks Vincent for confirming that the new code works with CubicSDR under Windows (and I am really glad it wasn't a hardware problem with the RSP2!)

Tonight I ran several tests using `SoapySDRServer` (SoapyRemote), connecting to it both with the command line test utility `SoapySDRUtil`, and with `CubicSDR`; I streamed from the RSPduo in single tuner, master, and slave modes (and combination of them), and there were no crashes or freezes (except in the case when I accidentally stopped the master before the slave, but that was my own fault, and I should have known better).

In the next few days I plan to keep running more tests (for instance using `gqrx`); this should also give @SDRplay, @nmaster2042, and other brave early testers the chance to try things out to make sure there are no major bugs before I merge it into the master branch.

Franco

nmaster2042, 2021-03-26:
@fventuri 

Sorry, I will make tests this week end.
I'll post update here.

[13 more replies]

closed 2021-04-10 by fventuri