pothosware/SoapySDRPlay3 #28: link error on macos catalina

BG7JAF, 2021-01-07:
I have install sdrplay api v3, and then make. compile is ok, but error at linking:

[ 16%] Linking CXX shared module libsdrPlaySupport.so
Undefined symbols for architecture x86_64:
  "SoapySDR::Device::setReferenceClockRate(double)", referenced from:
      vtable for SoapySDRPlay in Settings.cpp.o
  "SoapySDR::Device::getNativeDeviceHandle() const", referenced from:
      vtable for SoapySDRPlay in Settings.cpp.o
  "SoapySDR::Device::getReferenceClockRate() const", referenced from:
      vtable for SoapySDRPlay in Settings.cpp.o
  "SoapySDR::Device::getReferenceClockRates() const", referenced from:
      vtable for SoapySDRPlay in Settings.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libsdrPlaySupport.so] Error 1
make[1]: *** [CMakeFiles/sdrPlaySupport.dir/all] Error 2
make: *** [all] Error 2

what should I do?

fventuri, 2021-01-07:
@BG7JAF - I don't have a Mac here so my instructions might be slightly off, but it looks like those symbols are defined in SoapySDR version 0.8, but *not* in SoapySDR version 0.7, so I would check to make sure you have the correct version of SoapySDR installed there.

Here on my Linux system, one quick way to check that is to look at what the shared library `libSoapySDR.so` points to; for instance, here you could use `ls -l` like this:
```
$ ls -l /usr/local/lib64/libSoapySDR.so
lrwxrwxrwx. 1 root root 18 Dec 15  2019 /usr/local/lib64/libSoapySDR.so -> libSoapySDR.so.0.8
```
If, instead of libSoapySDR.so.0.8 you see a different version number, that might be what is causing that error.

Hope it helps,
Franco

guruofquality, 2021-01-07:
SoapySDRPlay3 should be compatible with the latest release of SoapySDR. This looks like a classic case of multiple installs of SoapySDR runtime files and development sources. Looks like it found master branch headers but the development symlink (.so) points to a release version. (just an approximate guess)

BG7JAF, 2021-01-11:
@fventuri I have check that the soapysdr version I install is 0.8:

➜  ~ ls /usr/local/lib/libSoapySDR*
/usr/local/lib/libSoapySDR.0.8.0.dylib /usr/local/lib/libSoapySDR.dylib
/usr/local/lib/libSoapySDR.0.8.dylib

@guruofquality I have not install other version libSoapySDR, the 0.8 version is install from source. It's quite strange.

fventuri, 2021-01-12:
@BG7JAF - I agree; what you are seeing is very strange and puzzling.

I'd like you to run a few more commands on your Mac to try to understand what's happening there.

First, please try to rebuild that SoapySDRPlay3 driver using `make VERBOSE=1`; this command should display the exact command that your Mac is running to link the compiled object files into the final shared library, and we should be able to see if there is any inconsistency there.

Then please run this command (or the equivalent for your Mac, since I am not sure if the command 'nm' works the same way for you); I am also showing you the first few lines of output on my Linux computer, so you have a better idea of what we are looking for:
```
$ nm -C /usr/local/lib64/libSoapySDR.so | grep ' T ' | grep 'SoapySDR::Device::' | sort -k3

000000000001cb8e T SoapySDR::Device::acquireReadBuffer(SoapySDR::Stream*, unsigned long&, void const**, int&, long long&, long)
000000000001cbc6 T SoapySDR::Device::acquireWriteBuffer(SoapySDR::Stream*, unsigned long&, void**, long)
000000000001ca9e T SoapySDR::Device::activateStream(SoapySDR::Stream*, int, long long, unsigned long)
000000000001ca7a T SoapySDR::Device::closeStream(SoapySDR::Stream*)
``` 

What I would like to know here is if that command displays any output (I hope so), and if you see any mention of say 'getNativeDeviceHandle' in the 120 or so output lines from that command; I'd like to see if that function (which is one of the missing ones in your initial post) is present or not in that output.

Franco

BG7JAF, 2021-01-14:
@fventuri Hi Franco, sorry for late to reply. Just now I recompile libsoapysdr, and install the missing dependency package doxygen which I don't think will affect the result, but to my surprise, all the errors are gone! XD seems it's my fault. anyway, thx very much!

closed 2021-01-14 by BG7JAF