pothosware/SoapySDRPlay3 #29: Multiple clients with SoapyRemote

pausrn, 2021-01-07:
Launching SoapyServer
`SoapySDRServer --bind`

Then connecting with a client (ex cubic sdr, rx_tools,...) works perfectly.

However trying to connect a second client is unsucceful : 

- Cubic SDR return no devices

- rx_tools return `
  [WARNING] findSDRPlay() called while the device is streaming. Ignoring request.
  SoapySDRDevice_make failed
  Failed to open sdr device matching 'driver=remote,remote=tcp://[::]:55132'.`

- SoapySDRUtil --find="driver=remote,remote:driver=sdrplay" return
  `No devices found! driver=remote,remote:driver=sdrplay`

This seems to be the default behaviour as per issue https://github.com/pothosware/SoapySDRPlay3/issues/9

However multiple consumer on one server seem to be possible on SoapyRemote and seemed to be possible in past (seeing issue [#46](https://webcache.googleusercontent.com/search?q=cache:yQY69vf0tG8J:https://github.com/pothosware/SoapySDRPlay/issues/46+&cd=1&hl=fr&ct=clnk&gl=us&client=firefox-b-d) on the SoapySDRPlay repository)

Trying to remove this line in Registration.cpp : 
`if (activeStream){
      SoapySDR_log(SOAPY_SDR_WARNING, "findSDRPlay() called while the device is streaming. Ignoring request.");
      return results;
 }`

Allow a device in use to be discovered from an other client, but give a segfault on connect.

Is this a limitation from the sdrplay driver? Have I misinterpreted the SoapyRemote page, and multiple client is not a feature?

fventuri, 2021-01-08:
Thanks for your feedback @pausrn 

Reading your comment, I assume you are using the 'master' branch of this SoapySDRPlay3 driver; would you mind trying using the branch called `support_for_serial_keyword` (https://github.com/pothosware/SoapySDRPlay3/tree/support_for_serial_keyword) instead?

We (mostly @nmaster2042 actually) already ran several tests using SoapySDRServer with the code in this new branch (see the comments in this thread: https://github.com/pothosware/SoapySDRPlay3/issues/23), and I plan to merge this branch into master, once I have addressed a few problems of crashes and client application freezing (hopefully this weekend, since on weekdays I can look at these issues only after my work).

Thanks for providing the useful command line test with `rx_tools`, so we can test my code fixes in more ways.

Franco

pausrn, 2021-01-08:
Sorry I did not know this was already worked on.

I tried the `support_for_serial_keyword` branch, detection of the rsp works flawlessly. But as you said, opening from a second instance will make the first stream hang.

Closing the issue as it is already worked on in https://github.com/pothosware/SoapySDRPlay3/issues/23

closed 2021-01-08 by pausrn