pothosware/SoapySDRPlay3 #30: change in windows registry path

guruofquality, 2021-01-12:
The build is looking for `GET_FILENAME_COMPONENT(SDRPLAY_API_DIR "[HKEY_LOCAL_MACHINE\\SOFTWARE\\SDRplay\\API;Install_Dir]" ABSOLUTE)`: https://github.com/pothosware/SoapySDRPlay3/blob/master/FindLibSDRplay.cmake

But the installer is placing the registry path in `[HKEY_LOCAL_MACHINE\\SOFTWARE\\SDRplay\\Service\\API` with the Service subdirectory in there

I think this should be change, I just want to make sure this is intentional going forward

SDRplay, 2021-01-12:
Confused what this is? API 3 always goes in SDRplay\Service\API - API 2 went into SDRplay\API

Is that a problem??

guruofquality, 2021-01-12:
Not a problem, Our current FindLibSDRplay.cmake in this repo is just incorrect and I wanted to confirm before just changing it

fventuri, 2021-01-12:
Thank @guruofquality and @SDRplay - I just fixed `FindLibSDRplay.cmake` and pushed it to master

Franco

closed 2021-02-07 by guruofquality