pothosware/SoapySDRPlay3 #31: Uninitialized outputSampleRate for RSP1a

rmoravcik, 2021-01-13:
Hello everyone,

I just received my RSP1a this week and it's working fine on my Ubuntu 20.10 machine. But GQRX is crashing on my RPi3 B+ setup.

I'm using the latest Raspberry Pi OS with self compiled gnuradio-3.8.1, the latest SoapySDRPlay3 and GQRX.

It looks like GQRX is trying to get sample rate during selection of new input device (https://github.com/csete/gqrx/blob/07722f21d0be2b63ae4e5d6f34601ece8625afe6/src/applications/gqrx/receiver.cpp#L221) but SoapySDRPlay3 is returning some uninitialized value different from 0, so GQRX will try to set back this value as sample rate.

And this will cause an crash, because it's not supported sample rate value.

Initializing the value of outputSampleRate variable (for example to 2MHz) will fix the issue for me. Something similar is already done for RSPDuo here: https://github.com/pothosware/SoapySDRPlay3/blob/208f95d117d3bce7359875646a99024a247ff39b/Settings.cpp#L163.

Thank you,
Roman

fventuri, 2021-01-13:
Thanks for your feedback @rmoravcik 

The issue you are seeing should be addressed by this commit on the `support_for_serial_keyword` branch: https://github.com/pothosware/SoapySDRPlay3/commit/e2ec2136cfd94cafeeab817094f5a5dd31bdff02

Since that branch has several fixes over the master branch, I'd suggest you give it a try with your RSP - you can find it here: https://github.com/pothosware/SoapySDRPlay3/tree/support_for_serial_keyword

I plan to merge the code from that branch into master as soon as some of the lingering issues with crashes and freezes are solved; see the full discussion here: https://github.com/pothosware/SoapySDRPlay3/issues/23

Franco

rmoravcik, 2021-01-14:
Thank you for your answer @fventuri. That fix for GQRX is exactly solving my problem.

Roman