pothosware/SoapySDRPlay3 #32: Device Not Detected

wmszeliga, 2021-03-02:
Well, that's the conclusion, although something is detected. 
I have an SDRplay RSP1A with the Soapy modules v0.8 from SDRplay and I'm using CubicSDR 0.2.6a (a build also from SDRplay) along with the SDRplay provided v0.8 Soapy modules. All of this is on a Mac running 10.15.7. Probing the device with SoapySDRUtil yields the following output
```
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] devIdx: 0
[INFO] hwVer: 255

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=20060DD397
  sdrplay_api_api_version=0.000000
  sdrplay_api_hw_version=255

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8, 9)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]
     * DabNotch Enable - DAB Notch Filter Control
       [key=dabnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 48] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 9] dB
  Full freq range: [0.001, 2000] MHz
    RF freq range: [0.001, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.0625, 0.125, 0.25, 0.5, 1, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz
```
As you can see, there are "Other Settings" available such as RF Gain Select and BiasT enable/disable. However, CubicSDR doesn't seem to fully see this device as evidenced by the lack of set/enable/disable options in the Settings menu for these "Other Settings". Below are some screenshots of the Settings menu and the initial dialog from CubicSDR. I will note that the radio does actually work in CubicSDR and I am able to receive MW AM broadcasts as well as VHF FM broadcasts. This issue has been discussed [here](https://github.com/cjcliffe/CubicSDR/issues/878)
[image]
[image]

fventuri, 2021-03-04:
@wmszeliga 
I don't have a Mac here, but I tried on my Linux (Fedora 33, to be exact) PC,  and I do see those additional settings (see screenshot below).

I pulled the current 'SoapySDRPlay3' code from the master branch, compiled, and installed it.

After that, I ran the command `SoapySDRUtil --probe=sdrplay`, and this was the output (I have an RSP2 so some of the values are slightly different from yours):

```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device sdrplay
[INFO] devIdx: 0
[INFO] hwVer: 2

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=**************
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=2

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * ExtRef Enable - External Reference Control
       [key=extref_ctrl, default=true, type=bool]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: Antenna A, Antenna B, Hi-Z
  Corrections: DC removal
  Full gain range: [0, 47] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 8] dB
  Full freq range: [0.001, 2000] MHz
    RF freq range: [0.001, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.0625, 0.096, 0.125, 0.192, 0.25, ..., 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz
```

Finally I started CubicSDR (version 0.2.6a), and when I move the cursor to the 'Settings' menu, I do see options like 'RF Gain Select' and 'BiasT Enable':

[image]

Franco

wmszeliga, 2021-03-04:
Curious. I see that my API version, as reported by SoapySDRUtil is `sdrplay_api_api_version=0.00000` while yours is ` sdrplay_api_api_version=3.070000`. Could this be indicative of an issue with the Mac-specific compilation of `libsdrPlaySupport.so` as compared with the Linux version?

fventuri, 2021-03-04:
@wmszeliga - good catch about the `sdrplay_api_api_version=0.00000`.

Let's first make sure our SDRplay setup and libraries match as closely as possible:

- I am using the SDRplay API version 3.07 downloaded from the SDRplay website; here on Linux I have it under `/usr/local/lib64`:
```
$ ls -l /usr/local/lib64/libsdrplay_api.so*

lrwxrwxrwx. 1 root root    19 May 24  2020 /usr/local/lib64/libsdrplay_api.so -> libsdrplay_api.so.3
lrwxrwxrwx. 1 root root    22 May 24  2020 /usr/local/lib64/libsdrplay_api.so.3 -> libsdrplay_api.so.3.07
-rw-r--r--. 1 root root 56176 Mar 22  2020 /usr/local/lib64/libsdrplay_api.so.3.07
```

- I started the service `sdrplay_api`; you should be able to see it running with `ps`:
```
$ ps -ef | grep sdrplay_api

root        2117       1  2 18:27 ?        00:02:55 /usr/local/bin/sdrplay_apiService
```

- My `libsdrPlaySupport.so` is linked to the library `libsdrplay_api.so.3.07` shown above; on Linux I can see that with the command 'ldd':
```
$ ldd /usr/local/lib64/SoapySDR/modules0.8/libsdrPlaySupport.so | grep libsdrplay_api

	libsdrplay_api.so.3.07 => /usr/local/lib64/libsdrplay_api.so.3.07 (0x00007fc0db254000)
```

I think the equivalent of `ldd` for MacOS is `otool -L`.

Franco

wmszeliga, 2021-03-04:
I too am using the SDRplay API version 3.0.7 from the SDRplay website:
```
$ ls -l /usr/local/lib/libsdrplay_api.so*
lrw-r--r--  1 root      wheel     34 Feb 19 17:57 /usr/local/lib/libsdrplay_api.so -> /usr/local/lib/libsdrplay_api.so.3
lrw-r--r--  1 root      wheel     37 Feb 19 17:57 /usr/local/lib/libsdrplay_api.so.3 -> /usr/local/lib/libsdrplay_api.so.3.07
-rwxr-xr-x  1 root      wheel  68308 May  8  2020 /usr/local/lib/libsdrplay_api.so.3.07
```
The sdrplay service is running
```
$ ps -ef | grep sdrplay_api
    0  5747     1   0  9:54PM ??         1:13.47 /usr/local/bin/sdrplay_apiService
```
and the sdrPlaySupport module is linked to the `libsdrplay_api.so.3.0.7` although I see that `otool` isn't reporting it with the full path. I wonder if that could be the issue.
```
$ otool -L libsdrPlaySupport.so 
libsdrPlaySupport.so:
	libsdrplay_api.so.3.07 (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libSoapySDR.0.8.dylib (compatibility version 0.8.0, current version 0.8.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
```

fventuri, 2021-03-04:
@wmszeliga 

I just re-read your initial post, and I see you are using the "SDRplay provided v0.8 Soapy modules", while in my case I built the SoapySDRPlay3 driver directly from the source code; perhaps the file you downloaded was built on a different version of MacOS (or has some other subtle difference), and does not work as expected.

If that is the case, would you mind downloading the source code from the master branch using `git clone`, and then building and installing it (the detailed instructions are here: https://github.com/pothosware/SoapySDRPlay3/wiki#building-soapy-sdr-play), to see if it makes any difference when you run CubicSDR?

Franco

wmszeliga, 2021-03-04:
Thanks, building SoapySDR from origin/master at 35189 and SoapySDRPlay3 from origin/master at 208f95, then copying libSoapySDR.0.8.dylib into my CubicSDR application bundle made the `sdrplay_api_api_version=3.070000` and the "Other Settings" now appear in the settings menu. Funny how the RSP1A basically worked without throwing any errors in any logs, but also subtly failed.

closed 2021-03-04 by wmszeliga