pothosware/SoapySDRPlay3 #37: Module hangs SoapySDR

ki4mcw, 2021-06-15:
Hey folks, having some trouble with this module. Running a find or a probe from SoapySDRUtil while the SDRPlay module is =present= in /usr/local/lib/SoapySDR/modules0.x, even if find-ing or probe-ing against another driver, causes SoapySDRUtil to hang. Simply doing a "sudo rm libsdrPlaySupport.so; sudo ldconfig" restores functionality.

Ubuntu 21.04:
Linux lenny-bu 5.11.0-18-generic #19-Ubuntu SMP Fri May 7 14:22:03 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

SoapySDR 0.7.2 (either from binary package, or compiled from source package)
-- OR 0.8 (compiled from latest source).

SoapySDRPlay3 (compiled from latest source, recompiled to switch between Soapy 0.7.2 and 0.8).

The msi* kernel drivers are blacklisted.
This same constellation (using Soapy 0.8 compiled from source) worked fine on this same PC under Ubuntu 20.10/64.

================================================
Example 1 - Attempting to use SDRPlay driver, with both
RTL-SDR and SDRPlay RSP1 hardware installed (any 
combination with/without hardware yields the same behavior,
though the kernel and Rafael 820 message only appear if
the RTL-SDR in plugged in)
================================================

$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.2-unknown
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/librtlsdrSupport.so  (0.3.0)
Module found: /usr/local/lib/SoapySDR/modules0.7/libsdrPlaySupport.so (0.3.0)
Available factories... rtlsdr, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

$ SoapySDRUtil --check=sdrplay
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Loading modules... done
Checking driver 'sdrplay'... PRESENT

$ SoapySDRUtil --probe=sdrplay
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device sdrplay
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver

<HANGS>

===============================================
Example 2 - Trying to use rtlsdr driver while sdrplay driver 
is merely =present= in modules dir, with both RTL-SDR 
and SDRPlay RSP1 hardware installed (again, the kernel
and Rafael 820 messages only appear if the RTL-SDR is
plugged in
===============================================

$ SoapySDRUtil --find="device=rtlsdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver

<HANGS>

===============================================
Example 3 - Delete sdrplay driver from modules dir and re-try,
RTL-SDR and SDRPlay RSP1 hardware installed
===============================================

$ cd /usr/local/lib/SoapySDR/modules0.7
$ ls -l
total 320
-rw-r--r-- 1 root root 143312 Jun 15 16:25 librtlsdrSupport.so
-rw-r--r-- 1 root root 183304 Jun 15 16:30 libsdrPlaySupport.so
$ sudo rm libsdrPlaySupport.so
$ sudo ldconfig
$ cd

$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.2-unknown
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/librtlsdrSupport.so (0.3.0)
Available factories... rtlsdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

$ SoapySDRUtil --probe=rtlsdr
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device rtlsdr
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Detached kernel driver
Found Rafael Micro R820T tuner

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=RTLSDR
  hardware=R820T
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * Direct Sampling - RTL-SDR Direct Sampling Mode
       [key=direct_samp, default=0, type=string, options=(0, 1, 2)]
     * Offset Tune - RTL-SDR Offset Tuning Mode
       [key=offset_tune, default=false, type=bool]
     * I/Q Swap - RTL-SDR I/Q Swap Mode
       [key=iq_swap, default=false, type=bool]
     * Digital AGC - RTL-SDR digital AGC Mode
       [key=digital_agc, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS8, CS16, CF32
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Size - Number of bytes per buffer, multiples of 512 only.
       [key=bufflen, units=bytes, default=262144, type=int]
     * Ring buffers - Number of buffers in the ring.
       [key=buffers, units=buffers, default=15, type=int]
     * Async buffers - Number of async usb buffers (advanced).
       [key=asyncBuffs, units=buffers, default=0, type=int]
  Antennas: RX
  Full gain range: [0, 49.6] dB
    TUNER gain range: [0, 49.6] dB
  Full freq range: [23.999, 1764] MHz
    RF freq range: [24, 1764] MHz
    CORR freq range: [-0.001, 0.001] MHz
  Sample rates: 0.25, 1.024, 1.536, 1.792, 1.92, 2.048, 2.16, 2.56, 2.88, 3.2 MSps

Reattached kernel driver

=====================================================
End examples
=====================================================

file:///home/rob/src2/soapy/SoapySDRPlay3-master/build/CMakeFiles/CMakeOutput.log

fventuri, 2021-06-15:
@ki4mcw - thanks for the reporting the problem and the detailed analysis.

In your description of the problem, I couldn't find if you are running or not the `sdrplay_api` service at the time of the tests (or perhaps I missed that).
To rule out that that the module hangs you are experiencing are related to that service, do you mind stopping it with the command:
```
systemctl stop sdrplay_api.service
```
then making sure that it is completely down with the commands:
```
ps -ef | grep sdrplay_api
ls /dev/shm
```
and finally re-running the command `SoapySDRUtil --find` with the `libsdrPlaySupport.so` module (the latest version, if possible) present and enabled - with this configuration I expect you see an output like this (which is what I see here when the service is down):
```
SoapySDRUtil --find 
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

shm_open: No such file or directory
[ERROR] sdrplay_api_Open() Error: sdrplay_api_Fail
[ERROR] Please check the sdrplay_api service to make sure it is up. If it is up, please restart it.
[ERROR] SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed
No devices found! 
```

but int this case it definitely should not hang or cause other crashes.

Based on what you see there, we can then decide what to look at next.

Franco

fventuri, 2021-06-15:
@ki4mcw - also to validate the SDRplay API library that the compiled SoapySDR module is going to use, do you mind running the `ldd` command on the installed module as follows to make sure it looks similar to what I see here on my computer?
```
$ ldd /usr/local/lib64/SoapySDR/modules0.8/libsdrPlaySupport.so 

	linux-vdso.so.1 (0x00007ffef97db000)
	libsdrplay_api.so.3.07 => /usr/local/lib64/libsdrplay_api.so.3.07 (0x00007f85c752a000)
	libSoapySDR.so.0.8 => /usr/local/lib64/libSoapySDR.so.0.8 (0x00007f85c748e000)
	libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007f85c726f000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f85c712b000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f85c7110000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f85c70ef000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f85c6f1e000)
	librt.so.1 => /lib64/librt.so.1 (0x00007f85c6f13000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007f85c6f0c000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f85c7786000)
```

Thanks,
Franco

ki4mcw, 2021-06-16:
My apologies Franco, multiple reboots and uninstall/re-install cycles somehow fixed this. Sorry to have wasted your time. 

Cheers,
Rob KI4MCW

fventuri, 2021-06-16:
@ki4mcw - Rob, I am glad to hear you were able to solve the problem with the SoapySDR module.

73,
Franco - K4VZ

closed 2021-06-16 by fventuri