pothosware/SoapySDRPlay3 #39: cmake version error

zekewup, 2021-06-22:
cmake ..
CMake Error at CMakeLists.txt:4 (cmake_minimum_required):
  CMake 3.13 or higher is required.  You are running version 3.10.2

fventuri, 2021-06-22:
@zekewup - thanks for reporting the problem with the minimum required version for cmake.

I checked what the other SoapySDR modules use for minimum required version for cmake, and it looks like most of them set it to 2.8.7; since I noticed that with a value that low cmake sends out some sort of future compatibility warning, I set it to 2.8.12 instead.

I just ran a full build from scratch here, and everything looks good to me, so I pushed that change to a new branch called `cmake_2_8_12` (https://github.com/pothosware/SoapySDRPlay3/tree/cmake_2_8_12); please give it a try to build it and run it, and if it works for you, I'll go ahead and merge that change to `CMakeLists.txt` into the `master` branch.

Franco

zekewup, 2021-06-23:
Franco 
Thank you i was now able to finish the install, However this must be a version not compatable with my SDRPLay Pro2 Can you do a this for the sdrplay2 version as well as just sdrplay? I have been trying to install them for 3 days and get this same error on all of them. Guess I will need to now un install this version ?

fventuri, 2021-06-23:
@zekewup
To be able to use this SoapySDRPlay driver you have to run SDRplay API version 3.07 (not the older 2.13 version).
You can follow the instructions to download it from this link: https://www.sdrplay.com/downloads/.
Once you have it installed and running, it should work with any of the RSP models, including the RSP2pro.

Franco

zekewup, 2021-06-27:
Franco, 

That fixed everything. now likes cmake version I have and Cubic is working great interfacing with my Ten Tec Orion etc Thank you!!!

fventuri, 2021-06-27:
@zekewup - glad to hear everything is working there

I just merged that branch (that lowers the minimum required version for cmake to 2.8.12) into the `master`branch.

73,
Franco

closed 2021-06-27 by fventuri