pothosware/SoapySDRPlay3 #44: libsdrPlaySupport.so cannot find libsdrplay_api.so.3.07

AsciiWolf, 2021-09-02:
Fedora 34, SoapySDR-0.7.2-11.fc34.x86_64, SDRplay API 3.07.1, latest SoapySDRPlay3 master.

The installed libsdrPlaySupport.so cannot find libsdrplay_api.so.3.07 although it linked fine during the build process:
```
$ ldd /usr/local/lib64/SoapySDR/modules0.7/libsdrPlaySupport.so                                                                                                                                                           
	linux-vdso.so.1 (0x00007ffd893cb000)
	libsdrplay_api.so.3.07 => not found
	libSoapySDR.so.0.7 => /lib64/libSoapySDR.so.0.7 (0x00007f8688a5e000)
	libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007f868883f000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f86886fb000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f86886e0000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f86886bf000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f86884ee000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007f86884e7000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f8688aec000)
```

```
$ SoapySDRUtil --probe="driver=sdrplay"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay
[ERROR] SoapySDR::loadModule(/usr/local/lib64/SoapySDR/modules0.7/libsdrPlaySupport.so)
  dlopen() failed: libsdrplay_api.so.3.07: cannot open shared object file: No such file or directory
```

However, the built libsdrPlaySupport.so in build dir (before installing it with `sudo make install`) does not have this issue:
```
$ ldd SoapySDRPlay3/build/libsdrPlaySupport.so
	linux-vdso.so.1 (0x00007ffe6c14d000)
	libsdrplay_api.so.3.07 => /usr/local/lib/libsdrplay_api.so.3.07 (0x00007f8ec52d1000)
	libSoapySDR.so.0.7 => /lib64/libSoapySDR.so.0.7 (0x00007f8ec5269000)
	libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007f8ec504a000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f8ec4f06000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f8ec4eeb000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f8ec4eca000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f8ec4cf9000)
	librt.so.1 => /lib64/librt.so.1 (0x00007f8ec4cee000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007f8ec4ce7000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f8ec5506000)
```

Everything works fine after manually copying this file to the target directory:
`sudo cp SoapySDRPlay3/build/libsdrPlaySupport.so /usr/local/lib64/SoapySDR/modules0.7/libsdrPlaySupport.so`

fventuri, 2021-09-03:
@AsciiWolf - I would run the following command first thing to make sure that the `ld` cache is refreshed:
```
sudo ldconfig
```
If you still keep having the same problem after the command above, please run the following command to make sure that `/usr/local/lib/libsdrplay_api.so.3.07` is in the `ld` cache:
```
ldconfig -p | grep sdrplay
```

Another thing I noticed is that you seem to be running an older version of SoapySDR (version 0.7); perhaps you may also want to git clone and rebuild SoapySDR using the latest version (0.8) - I am on Fedora 34 too, and SoapySDR version 0.8 works without any problems here.

Hope this helps,
Franco

AsciiWolf, 2022-04-12:
Turns out it was a problem with Fedora not having /usr/local/lib in its ld library path by default. Was fixed by creating `/etc/ld.so.conf.d/local.conf` with this content:

```
/usr/local/lib
/usr/local/lib64
```

closed 2022-04-12 by AsciiWolf

rstrawsb, 2023-02-05:
This worked for me as well