pothosware/SoapySDRPlay3 #47: Could not find a package configuration file provided by "SoapySDR"

tomfanning, 2021-12-19:
Me again.

Following the [build guide](https://github.com/pothosware/SoapySDRPlay3/wiki#building-soapy-sdr-play) on latest Raspberry Pi OS, after the [eventually successful SoapySDR install](https://github.com/pothosware/SoapySDR/issues/335):

```
pi@sdrpi:~/SoapySDRPlay3/build $ cmake ..
CMake Error at CMakeLists.txt:7 (find_package):
  Could not find a package configuration file provided by "SoapySDR"
  (requested version 0.4.0) with any of the following names:

    SoapySDRConfig.cmake
    soapysdr-config.cmake

  Add the installation prefix of "SoapySDR" to CMAKE_PREFIX_PATH or set
  "SoapySDR_DIR" to a directory containing one of the above files.  If
  "SoapySDR" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/pi/SoapySDRPlay3/build/CMakeFiles/CMakeOutput.log".
pi@sdrpi:~/SoapySDRPlay3/build $ cmake .. -DSoapySDR_DIR=/home/pi/SoapySDR/cmake/Modules
CMake Error at CMakeLists.txt:7 (find_package):
  Could not find a configuration file for package "SoapySDR" that is
  compatible with requested version "0.4.0".

  The following configuration files were considered but not accepted:

    /home/pi/SoapySDR/cmake/Modules/SoapySDRConfig.cmake, version: unknown

-- Configuring incomplete, errors occurred!
See also "/home/pi/SoapySDRPlay3/build/CMakeFiles/CMakeOutput.log".
```

guruofquality, 2021-12-19:
Can you share the output of make install for SoapySDR? Usually cmake files are installed to /usr/local/share/cmake/SoapySDR/ and you should not have to specify -DSoapySDR_DIR for default paths like that

tomfanning, 2021-12-19:
```
pi@sdrpi:~/SoapySDR/build $ sudo make install
[ 60%] Built target SoapySDR
[ 71%] Built target SoapySDRUtil
[ 76%] Built target TestKwargsMarkup
[ 81%] Built target TestFormatParser
[ 86%] Built target TestConvertTypes
[ 92%] Built target TestTimeConversion
[ 94%] Built target SoapySDR_swig_compilation
[100%] Built target _SoapySDR
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRConfig.cmake
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRUtil.cmake
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRConfigVersion.cmake
-- Up-to-date: /usr/local/include/SoapySDR
-- Up-to-date: /usr/local/include/SoapySDR/Time.h
-- Up-to-date: /usr/local/include/SoapySDR/Constants.h
-- Up-to-date: /usr/local/include/SoapySDR/Time.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Errors.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Config.h
-- Up-to-date: /usr/local/include/SoapySDR/Modules.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Device.h
-- Up-to-date: /usr/local/include/SoapySDR/ConverterPrimitives.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Version.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Types.h
-- Up-to-date: /usr/local/include/SoapySDR/Types.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Errors.h
-- Up-to-date: /usr/local/include/SoapySDR/Formats.h
-- Up-to-date: /usr/local/include/SoapySDR/Logger.h
-- Up-to-date: /usr/local/include/SoapySDR/Version.h
-- Up-to-date: /usr/local/include/SoapySDR/Device.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Logger.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Formats.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Registry.hpp
-- Up-to-date: /usr/local/include/SoapySDR/ConverterRegistry.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Converters.h
-- Up-to-date: /usr/local/include/SoapySDR/Config.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Modules.h
-- Up-to-date: /usr/local/lib/libSoapySDR.so.0.8.1
-- Up-to-date: /usr/local/lib/libSoapySDR.so.0.8
-- Up-to-date: /usr/local/lib/libSoapySDR.so
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRExport.cmake
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRExport-release.cmake
-- Up-to-date: /usr/local/lib/pkgconfig/SoapySDR.pc
-- Up-to-date: /usr/local/bin/SoapySDRUtil
-- Up-to-date: /usr/local/share/man/man1/SoapySDRUtil.1
-- Up-to-date: /usr/local/lib/python3.9/dist-packages/_SoapySDR.so
-- Up-to-date: /usr/local/lib/python3.9/dist-packages/SoapySDR.py
```

fventuri, 2021-12-19:
@tomfanning
please try rerunning the `cmake` that failed with the additional option `CMAKE_FIND_DEBUG_MODE=ON`, like this:
```
cmake -D CMAKE_FIND_DEBUG_MODE=ON ..
```
it should tell us why `cmake` looks under `/home/pi/SoapySDR` to find that config file.

Franco

tomfanning, 2021-12-19:
```
pi@sdrpi:~/SoapySDRPlay3/build $ cmake -D CMAKE_FIND_DEBUG_MODE=ON ..
CMake Debug Log at CMakeLists.txt:7 (find_package):
  <PackageName>_ROOT CMake variable [CMAKE_FIND_USE_PACKAGE_ROOT_PATH].

    none

  CMAKE_PREFIX_PATH variable [CMAKE_FIND_USE_CMAKE_PATH].

    none

  CMAKE_FRAMEWORK_PATH and CMAKE_APPBUNDLE_PATH variables
  [CMAKE_FIND_USE_CMAKE_PATH].

    none

  Env variable SoapySDR_DIR [CMAKE_FIND_USE_CMAKE_ENVIRONMENT_PATH].

    none

  CMAKE_PREFIX_PATH env variable [CMAKE_FIND_USE_CMAKE_ENVIRONMENT_PATH].

    none

  CMAKE_FRAMEWORK_PATH and CMAKE_APPBUNDLE_PATH env variables
  [CMAKE_FIND_USE_CMAKE_ENVIRONMENT_PATH].

    none

  Paths specified by the find_package HINTS option.

    none

  Standard system environment variables
  [CMAKE_FIND_USE_SYSTEM_ENVIRONMENT_PATH].

    /usr/local
    /usr
    /
    /usr/local/games
    /usr/games

  CMake User Package Registry [CMAKE_FIND_USE_PACKAGE_REGISTRY].

    none

  CMake variables defined in the Platform file
  [CMAKE_FIND_USE_CMAKE_SYSTEM_PATH].

    /usr/X11R6
    /usr/pkg
    /opt

  CMake System Package Registry
  [CMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY].

    none

  Paths specified by the find_package PATHS option.

    none

  find_package considered the following locations for the Config module:

    /usr/local/SoapySDRConfig.cmake
    /usr/local/soapysdr-config.cmake
    /usr/SoapySDRConfig.cmake
    /usr/soapysdr-config.cmake
    /SoapySDRConfig.cmake
    /soapysdr-config.cmake
    /usr/local/games/SoapySDRConfig.cmake
    /usr/local/games/soapysdr-config.cmake
    /usr/games/SoapySDRConfig.cmake
    /usr/games/soapysdr-config.cmake
    /opt/SoapySDRConfig.cmake
    /opt/soapysdr-config.cmake

  The file was not found.

CMake Error at CMakeLists.txt:7 (find_package):
  Could not find a package configuration file provided by "SoapySDR"
  (requested version 0.4.0) with any of the following names:

    SoapySDRConfig.cmake
    soapysdr-config.cmake

  Add the installation prefix of "SoapySDR" to CMAKE_PREFIX_PATH or set
  "SoapySDR_DIR" to a directory containing one of the above files.  If
  "SoapySDR" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/pi/SoapySDRPlay3/build/CMakeFiles/CMakeOutput.log".
```

fventuri, 2021-12-19:
@tomfanning 
very strange; this is what I see on my Raspberry Pi when I run the same command (I am showing only the relevant parts):

```
...

  find_package considered the following locations for the Config module:

    /usr/local/SoapySDRConfig.cmake
    /usr/local/soapysdr-config.cmake
    /usr/local/share/cmake/SoapySDR/SoapySDRConfig.cmake

  The file was found at

    /usr/local/share/cmake/SoapySDR/SoapySDRConfig.cmake

...
```

I am not sure why in your case it doesn't look under `/usr/local/share/cmake`, since it should be in `cmake` config mode search path according to the documentation here: https://cmake.org/cmake/help/latest/command/find_package.html#config-mode-search-procedure

What version of `cmake` are you running?
```
cmake --version
```

Franco

tomfanning, 2021-12-19:
```
pi@sdrpi:~ $ cmake --version
cmake version 3.18.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

This is a pretty fresh Raspberry Pi OS (lite, headless) installed just a few days ago, with very little done to it.

fventuri, 2021-12-19:
It's exactly the same version I am running here with the latest Raspberry OS.

Perhaps you have some environment variable with special meaning to `cmake` (see here: https://cmake.org/cmake/help/latest/manual/cmake-env-variables.7.html)?

```
set | grep CMAKE
```

Franco

```
$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

$ uname -a
Linux raspberrypi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux
```

tomfanning, 2021-12-19:
```
pi@sdrpi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@sdrpi:~ $ set | grep CMAKE
pi@sdrpi:
```

fventuri, 2021-12-19:
hmmm.... a directory/file permission problem perhaps?
```
ls -l /usr/local/share/cmake/SoapySDR
```
(run the above as user 'pi', without `sudo`)

Franco

tomfanning, 2021-12-19:
Bingo.
I installed wireguard, having run `umask 077` in this terminal window earlier, then forgot.

tomfanning, 2021-12-19:
Thanks so much for the sanity check / troubleshoot 🥇

closed 2021-12-19 by tomfanning

amitpathania, 2022-04-19:
I am also getting the same error while running cmake .. . I am using ubuntu18.
CMake Error at CMakeLists.txt:9 (find_package):
  Could not find a package configuration file provided by "SoapySDR"
  (requested version 0.4.0) with any of the following names:

    SoapySDRConfig.cmake
    soapysdr-config.cmake

  Add the installation prefix of "SoapySDR" to CMAKE_PREFIX_PATH or set
  "SoapySDR_DIR" to a directory containing one of the above files.  If
  "SoapySDR" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/lenevo/Documents/soapy-bb60/SoapyBB60C/build/CMakeFiles/CMakeOutput.log".

`
Version of cmake is
`cmake version 3.10.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
`
Please help me resolving the issue.

amitpathania, 2022-04-30:
Issue still not resolved. Getting following error while running this `cmake -D CMAKE_FIND_DEBUG_MODE=ON ..
`

`CMake Error: The source directory "/home/lenevo/Documents" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
`

fventuri, 2022-05-01:
@amitpathania - I am confused.

Which SoapySDR module are you trying to build? soapy-bb60 or SoapySDRPlay?

If it is the former, you may want to create a support request with the vendor or check their support forums (https://signalhound.com/support/forums/).
If you are try to build this SoapySDRPlay module, it looks like you cloned the wrong repository; in case I'd suggest to start over and follow the instructions in the Wiki (https://github.com/pothosware/SoapySDRPlay3/wiki#building-soapy-sdr-play).

Franco

amitpathania, 2022-05-03:
OK. Thanks. Just wanted to get help as error was same. Not to worry, will create separate issue there