pothosware/SoapySDRPlay3 #49: Unable to compile on Raspberry Pi4 with /usr/local/lib/libsdrplay_api.so: file not recognized: file format not recognized Error

gah67, 2022-02-05:
```
[ 16%] Linking CXX shared module libsdrPlaySupport.so
/usr/local/lib/libsdrplay_api.so: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/sdrPlaySupport.dir/build.make:165: libsdrPlaySupport.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:123: CMakeFiles/sdrPlaySupport.dir/all] Error 2
make: *** [Makefile:149: all] Error 2

pi@rpi4 ~/repos/SoapySDRPlay/build master 17s
❯ file /usr/local/lib/libsdrplay_api.so.3.07
/usr/local/lib/libsdrplay_api.so.3.07: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, BuildID[sha1]=3aea2df7ad2947cfa4a385f73779dca8f1eb3008, with debug_info, not stripped

```
SDRPlay api installed without errors using SDRplay_RSP_API-ARM64-3.07.1.run from SDRPlay
SoapySDR built from git

Hardware is RPi4 8GB with 64-bit kernel & 32-bit userland (which I am guessing causes the issue)

fventuri, 2022-02-05:
@gah67 - I think you are spot on.

I have a Raspberry Pi 4 here running 32bit Raspberry Pi OS, and I am using the `libsdrplay_api.so.3.07` shared library that comes from this download: https://www.sdrplay.com/software/SDRplay_RSP_API-ARM32-3.07.2.run

When I run the `file` command like you did, my output looks like this:
```
$ file /usr/local/lib/libsdrplay_api.so.3.07 
/usr/local/lib/libsdrplay_api.so.3.07: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, BuildID[sha1]=7f1aa4268ead12fe235ac3de2494c4691d9a76d7, stripped
```
Using this version of the SDRplay API shared library I was able to build and run this SoapySDRPlay3 module without any problems.

Franco

gah67, 2022-02-05:
@fventuri so I think I should have the 32-bit shared library as it is 32-bit userland BUT the SDRPlay installer refuses to install the 32-bit version as it does a kernel check but not userland check (if there is one - I am out of my depth here!!) before installing

Now that the full 64-bit Lite OS is available I think I will try that and hope that all of the other software works

fventuri, 2022-02-05:
@gah67 - I didn't know that the 64bit version of Raspberry OS came out a few days ago.
I'll grab an SD card, give it a try, and I'll let you know what I find out.

Franco

alphafox02, 2022-02-05:
DragonOS Pi64 is a 64bit OS for the pi4. I’ve had no problems using the SDRPlay api that’s included or this Soapy branch. I suspect you’ll find similar results with the raspbian OS that’s now 64bit.

gah67, 2022-02-06:
@fventuri @alphafox02 Yes, with a full 64-bit version of the Pi OS it builds first time

Thinking about this, having looked further into it, this is actually an issue with the api installer from SDRPlay.

This project does what it is supposed to.

Thanks for the comments & help

closed 2022-02-06 by gah67