pothosware/SoapySDRPlay3 #50: Question about rfnotch_ctrl

jaymot123, 2022-03-06:
The SoapySDRPlay3 module has two filter controls, rfnotch_ctrl and dabnotch_ctrl (or maybe these are in the API, not the Soapy module?) The DAB notch control is self-explanatory (except that I don't know what DAB is), but other than that notch filter my RSP1A has two others. for the medium-wave AM and the FM broadcast bands. Does enabling the RF notch via rfnotch_ctrl=1 enable both of these broadcast band filters? I looked at the API documentation but couldn't find an answer: in there it appears that there may be separate controls for the MW and FM filters. Thanks.

fventuri, 2022-03-06:
@jaymot123 - DAB stands for 'Digital Audio Broadcasting' (https://en.wikipedia.org/wiki/Digital_Audio_Broadcasting) and it is a digital broadcasting service used for commercial radio stations (similar to FM radio except that it is digital) in many countries. It uses the 174-239MHz frequency band.

As per the setting `rfnotch_ctrl` in the case of the RSP1A it sets the SDRplay API RSP1 parameter `rfNotchEnable` (https://github.com/pothosware/SoapySDRPlay3/blob/master/Settings.cpp#L1486) - the SDRplay API Specification Guide (https://www.sdrplay.com/docs/SDRplay_API_Specification_v3.07.pdf) on page 15 for the RSP1A lists only this 'RF Notch' and the 'DAB RF Notch' that you mentioned.

Since the datasheet for the RSP1A (https://www.sdrplay.com/docs/RSP1Adatasheetv1.9.pdf) shows three different notch filters (FM, MW, and DAB), probably the 'RF Notch' setting enables both the FM and MW filters, while the 'DAB RF Notch' filter controls the DAB filter.
In order to be 100% sure this is the case you may want to contact SDRplay technical support about this specific question.

Hope this helps,
Franco

jaymot123, 2022-03-07:
Thank you, Franco.

closed 2022-03-07 by jaymot123