pothosware/SoapySDRPlay3 #52: Master (RSPduo sample rate=8mhz)

righthalfplane, 2022-06-03:
I have that as an option in the list, but when I select it - it offers a maximum sample rate of 2 MHZ and if I ask it to do any sample rate above 2 MHZ - I complains a does not do it on macOS Monterey

SDRplay, 2022-06-03:
If you're in single tuner mode then 8 MHz is fine, but if you are dual tuner or master/slave mode, 2 MHz is the maximum sample rate for each tuner. Make sure you are selecting the single tuner device when the device selection appears.

Best regards,

Andy

fventuri, 2022-06-03:
It is important to notice that the RSPduo in Dual Tuner, Master, and Slave mode has actually two different sample rates that you need to be aware of.

The first one is the sample rate assigned to `rspDuoSampleFreq` (see include file `sdrplay_api.h` line 147) and to `fsHz` (`sdrplay_api_dev.h` line 19) and that value can be only 6MSps or 8Msps depending on the IF frequency; the second one is the actual output sample rate which is always 2Msps (before decimation), and this is the actual sample rate of the values as seen by the client application (i.e. the one that is actually used in the SDR program).

You can read more about LIF down-conversion on page 26 of the SDRplay API Specification guide (https://www.sdrplay.com/docs/SDRplay_API_Specification_v3.07.pdf)

Franco

righthalfplane, 2022-06-03:
Thanks for the response - I thought that it was another option where I could get 8 MSPS of the Master and I was wondering why it did not work.