pothosware/SoapySDRPlay3 #54: RSP1A not found under MAC OS after update from SoapySDRplay to SoapySDRplay3

aneitzel, 2022-08-27:
Hello, 

**RSP1A cant be found under MAC OS after update from SoapySDRplay to SoapySDRplay3 (OS is Catalina 1.15.7).
If i revert back to SoapySDRplay and SDRplay (package install all via MAC PORTs) it will be recognized and running fine.**

Also i noticed that RSP1A is after a fresh boot up only been seen then run "SoapySDRUtil --find" twice times (under old SoapySDRplay). I read in SDRplay Forum this is solved with Catalina OS, but seems its not. Anyway back to SoapySDRplay3 and RSP1A not found problem. I tested the following:

- a complete fresh install of MAC OS
- install SoapySDRplay3 and SDRplay3 + dependencies via MAC ports
- also i tested the install of the latest API package SDRplay_RSP_API-MacOSX-3.07.3.pkg from SDRplay Website

Then i verified that OS see my RSPA1 via lsusb:
alpha@iMAC ~ % uptime
11:06  up 53 secs, 2 users, load averages: 2,29 0,63 0,23
alpha@iMAC ~ % lsusb -v -d 1df7:3000
Bus 020 Device 003: ID 1df7:3000 SDRplay RSP1a
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0        64
  idVendor           0x1df7 SDRplay
  idProduct          0x3000 RSP1a
  bcdDevice            2.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           66
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              400mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      .....

alpha@iMAC ~ % SoapySDRUtil --probe=sdrplay
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################
Probe device sdrplay
Error probing device: SoapySDR::Device::make() no driver specified and no enumeration results  

alpha@iMAC ~ % SoapySDRUtil --find=sdrplay
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

^C
alpha@iMAC ~ % SoapySDRUtil --info        
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-unknown
API Version: v0.8.0
ABI Version: v0.8
Install root: /opt/local
Search path:  /opt/local/lib/SoapySDR/modules0.8
Module found: /opt/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so (0.4.0)
Available factories... sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

-----
alpha@iMAC ~ % ps aux |grep sdrplay
root               253   0,0  0,0  4301168   1868   ??  S    11:54am   0:00.09 /opt/local/bin/sdrplay_apiService
root               100   0,0  0,0  4456720   2728   ??  Ss   11:54am   0:00.01 /opt/local/bin/daemondo --label=sdrplay_service --start-cmd /opt/local/bin/sdrplay_apiService ; --pid=exec
alpha             1074   0,0  0,0  4408516    756 s016  S+    1:13pm   0:00.00 grep sdrplay
-----

As stated, then i revert to the old packages SDRplay + SoapySDRplay the RSP1A works as exspected.
Need your help to figure out whats going on here please

regards André

aneitzel, 2022-08-27:
After some play around SoapySDRUtil brings this error (but seems the API service is running, a stop/start dont helped)

alpha@iMAC ~ % SoapySDRUtil --find                             
######################################################
     Soapy SDR -- the SDR abstraction library     
######################################################

shm_open: No such file or directory
[ERROR] sdrplay_api_Open() Error: sdrplay_api_Fail
[ERROR] Please check the sdrplay_api service to make sure it is up. If it is up, please restart it.
[ERROR] SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed

alpha@iMAC ~ % ps aux |grep sdrplay                            
root               253   0,0  0,0  4301168   1868   ??  S    11:54am   0:00.10 /opt/local/bin/sdrplay_apiService
root               100   0,0  0,0  4456720   2728   ??  Ss   11:54am   0:00.01 /opt/local/bin/daemondo --label=sdrplay_service --start-cmd /opt/local/bin/sdrplay_apiService ; --pid=exec
alpha             1136   0,0  0,0  4399300    736 s016  S+    1:20pm   0:00.00 grep sdrplay
alpha@iMAC ~ %

fventuri, 2022-08-27:
@aneitzel - I think the right command to probe for a SDRplay device using SoapySDRUtil is:
```
SoapySDRUtil --probe="driver=sdrplay"
```
see: https://github.com/pothosware/SoapySDRPlay3/wiki#probing-soapy-sdr-play

Also I too have seen in the past cases where the SDRplay API service showed up as running, but for some reason wasn't responding.
To do a full clean the restart of the SDRplay API service, these are I steps I run here on Linux (I imagine they are probably very similar on Mac):

- stop the service (here on Linux I run `systemctl stop sdrplay.service`)
- after the previous command returns, I check with `ps` to see if the service if still running and if so, I'll stop it with `kill -TERM <pid>` (and when that doesn't work either, I'll terminate it with `kill -KILL <pid>`)
- also please note that tthe SDRplay API service sometime leaves some shared memory files under '/dev/shm' and if they are there, they need to be removed; I do that with something like this: `rm /dev/shm/Glbl\\sdrSrv*`
- finally I restart the SDRplay API service with `systemctl start sdrplay.service`

Of course rebooting takes care of all these steps, but I find this process faster (and less disruptive) that a full reboot.

Hope this helps,
Franco

aneitzel, 2022-08-27:
i found something out !

the RSP1A will be working then i UNLOAD the service startup created during installation with
"sudo port unload SDRplay3"

After check now running services:
alpha@iMAC ~ % ps aux |grep sdrplay
alpha              552   0,0  0,0  4399300    736 s000  S+    6:23pm   0:00.00 grep sdrplay
**root                87   0,0  0,0  4289888   1080   ??  Ss    6:22pm   0:00.04 /usr/local/bin/sdrplay_apiService**

>>>>>>>>>>> RSP1A is OK and working !!!!!!!

If the service startup will enabled again via "sudo port load SDRplay3" seems there are two instances of backgound API service running

**1. /usr/local/bin/sdrplay_apiService
2. /opt/local/bin/sdrplay_apiService**

see here:

alpha@iMAC ~ % sudo port load SDRplay3            
--->  Loading startupitem 'sdrplay_service' for SDRplay3

alpha@iMAC ~ % ps aux |grep sdrplay                 
root               663   0,0  0,0  4272496   1808   ??  S     6:26pm   0:00.01 /opt/local/bin/sdrplay_apiService
root               662   0,0  0,0  4427024   2620   ??  Ss    6:26pm   0:00.01 /opt/local/bin/daemondo --label=sdrplay_service --start-cmd /opt/local/bin/sdrplay_apiService ; --pid=exec
root                87   0,0  0,0  4604016   3592   ??  Ss    6:22pm   0:00.39 /usr/local/bin/sdrplay_apiService
alpha              665   0,0  0,0  4399300    736 s000  S+    6:26pm   0:00.00 grep sdrplay

IF THIS IS THE CASE, THE RSP1A will never found and Soapy probe hangs

Any idea tfrom there the 2. instance is and how to cleanup this problem?

fventuri, 2022-08-28:
@aneitzel - looking at your message I noticed that the two instances of the sdrplay_apiService are in two different locations: `/usr/local/bin` and `/opt/local/bin`.
This makes me think that you have two different installations of the SDRplay API on your Mac: one could come from running the SDRplay API installer provided by SDRplay (that one puts the executable of sdrplay_apiService under `/usr/local/bin`, at least here on Linux) and the other could come from Mac ports (I am not familiar with it, but that is what it looks like based on the `sudo port ...` commands in your comment).

Once you decide which of the two installations you want to keep, I suggest removing the other one to avoid conflicts like this one.
If you decide to remove the one under `/usr/local/bin` and keep the other other one, I think you have to stop and disable the service first (under Linux the commands are `systemctl stop sdrplay.service; systemctl disable sdrplay.service`; I imagine the commands are similar on your Mac), and then remove the executable from `/usr/local/bin`, since I don't think SDRplay provides an uninstall script.
If you decide instead to keep the instance under `/usr/local/bin` and remove the one from Mac ports, you can probably do that with a single `sudo port ...` command.

Franco

ra1nb0w, 2022-08-28:
You should remove the `/usr/local/` version installed manually with `pkg` to ensure API matching.
I don't know if upstream has a removal procedure or a script but you can start with `find /usr/local` then remove manually the files of sdrplay. Maybe check with `launchctl` if there is some related service but this point is not important. Reboot and you have a working device.

aneitzel, 2022-08-28:
Solution: **Disabling one of the API instances** 

If someone can provide instructions how to deinstall API package from SDRplay website? Would be very appreciated!

ra1nb0w, 2022-08-28:
Use `find` and then `rm` from terminal

aneitzel, 2022-08-28:
Guys, seems there is something wrong with the API comes with MAC PORTS. With this API active my RSP1A will not be found!

 I cleaned up the SDRplay webpage provided API via 
rm /Library/LaunchDaemons/com.sdrplay.sdrplay_service.plist
     /usr/local/bin
     /usr/local/include
     /usr/local/lib

+ enabling the API startup via MAC PORTS:

alpha@iMAC ~ % sudo port load SDRplay3                                                 
Password:
**--->  Loading startupitem 'sdrplay_service' for SDRplay3**

+ REBOOT
------------------

alpha@iMAC ~ % ps aux |grep sdrplay                 
root               253   0,0  0,0  4291952   1844   ??  S     1:40pm   0:00.04 /opt/local/bin/sdrplay_apiService
alpha              547   0,0  0,0  4408516    748 s004  S+    1:41pm   0:00.00 grep sdrplay
root                99   0,0  0,0  4454672   2676   ??  Ss    1:40pm   0:00.01 /opt/local/bin/daemondo --label=sdrplay_service --start-cmd /opt/local/bin/sdrplay_apiService ; --pid=exec

alpha@iMAC ~ % SoapySDRUtil --info                  
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-unknown
API Version: v0.8.0
ABI Version: v0.8
Install root: /opt/local
Search path:  /opt/local/lib/SoapySDR/modules0.8
Module found: /opt/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so (0.4.0)
Available factories... sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
alpha@iMAC ~ % SoapySDRUtil --probe="driver=sdrplay"
######################################################
     Soapy SDR -- the SDR abstraction library     
######################################################

Probe device driver=sdrplay
**Error probing device: no available RSP devices found**

alpha@iMAC ~ % SoapySDRUtil --find                  
######################################################
     Soapy SDR -- the SDR abstraction library     
######################################################

**No devices found!** 
alpha@iMAC ~ % 

Any idea?

ra1nb0w, 2022-08-28:
on a fresh macOS 10.14 I did `sudo port install soapysdrplay3`; installed everything from pre-compiled packages (1 second); reboot and everything worked fine with RSP1A