pothosware/SoapySDRPlay3 #65: Mac M1 Issue

madjuju, 2023-02-16:
[ 16%] Linking CXX shared module libsdrPlaySupport.so
Undefined symbols for architecture arm64:
  "SoapySDR::Device::getSettingInfo(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const", referenced from:
      vtable for SoapySDRPlay in Settings.cpp.o
  "SoapySDR::Device::getSettingInfo(int, unsigned long, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const", referenced from:
      vtable for SoapySDRPlay in Settings.cpp.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libsdrPlaySupport.so] Error 1
make[1]: *** [CMakeFiles/sdrPlaySupport.dir/all] Error 2
make: *** [all] Error 2

fventuri, 2023-02-17:
@madjuju - it looks like `libSoapySDR` on your Mac is missing that `getSettingInfo` function for some reason.

To confirm this is really the case, do you mind running the following command from a terminal:
```
find /usr -name libSoapySDR\*
``` 
this output should hopefully be the full path of the SoapySDR dynamic library on your system; probably something like this: `/usr/local/lib/libSoapySDR.<version>.dylib`

Once you have that filename, the next step is to run the command:
```
nm -C <filename from the previous step> | grep 'SoapySDR::Device::getSettingInfo'
```
and see if it returns anything.

I don't have a Mac here, since I run Linux on my computer; you may have to tweak the commands above a little bit in case you get some error message or warning.

Franco

Aldekein, 2023-06-08:
I have the same issue, but on M2. 
I have tried `SDRplay_RSP_API-ARM64-3.07.1.run`, `SDRplay_RSP_API-MacOS-3.07.a4.run`, `SDRplay_RSP_API-MacOSX-3.07.3.pkg` – the proper arm64 module, that should be placed in `lib/SoapySDR/modules0.8/libsdrPlaySupport.so` is absent :(