pothosware/SoapySDRPlay3 #67: Interfacing SoapySDRPlay with rx_tools (RSPDuo)

dhollosi, 2023-04-11:
Hi everyone,

Not sure if this is the right thread but I would appreciate any help on my issue! 

I have managed to acquire raw IQ data from the Tuner 1 (in Single Tuner mode) with the following command:

`./rx_sdr -d "driver=sdrplay" -a "Tuner 1 50 ohm" -f 1600M -s 2M -I CS16 -F CS16 -t rfgain_sel=0 test_iq.raw `

I was however hoping to be able to do the same with the Tuner 2 (the one that has the bias T) with the following command: 

`./rx_sdr -d "driver=sdrplay" -a "Tuner 2 50 ohm" -f 1600M -s 2M -I CS16 -F CS16 -t rfgain_sel=0,biasT_ctrl=true test2_iq.raw `

But the outputted data is unusable - I suspect it is still using Tuner 1 (which doesn't have my antenna connected to it). Strangely enough, the bias T still manages to power my LNA..

As an added benefit, I was hoping to use both tuners in Dual Tuner mode with rx_tools but after having got lost in the api documentation, I am unsure how to reach this goal (if possible).

Furthermore, I am struggling to understand how one can select the bandwidth given an inputted sample rate (e.g 2 MSPS).   

Kind regards,
Dimitri

dhollosi, 2023-04-13:
Hello Franco,

I tried with the latest pull but nothing has changed on my side. I even re-installed SoapySDR and SoapySDRPlay3 from scratch but no success yet.

Will look more into it this afternoon and get back to you.

Dimitri

fventuri, 2023-04-13:
Dimitri,
to make sure you are using the latest version from the `copy-device-and-tuner-config` branch, and that your problem too is caused by switching tuner from `Tuner 1` to `Tuner 2` like I saw here, can you please add this log line right before the changes I made (i.e. right before this line: https://github.com/pothosware/SoapySDRPlay3/blob/copy-device-and-tuner-config/Settings.cpp#L345):
```
    SoapySDR_log(SOAPY_SDR_INFO, "about to switch tuner - saving and then restoring settings");
```

Then rebuild and reinstall the SoapySDRPlay3 module; once you run `rx_sdr` and set it to use the antenna `Tuner 2 50 ohm`, you should see that info message in the terminal; if you don't, then we have to figure out what's different between my setup here and yours.

Franco

dhollosi, 2023-04-20:
Hi Franco,

I apologise for the lack of response, work got a bit hectic since the end of last week. 

I re-installed SoapySDRPlay3, checked out the new branch, built and installed it.

Thank you very much for your help, everything works like a charm now (even with the BiasT)! As far as I'm concerned, you could proceed with the merge to the main branch.

All the best,
Dimitri

fventuri, 2023-04-21:
Dimitri,
you are in good company about having a hectic week at work!
Anyway thanks for trying it out and making sure thst `rx_sdr` now works as expected; I just merged that change into the `master` branch.

Franco