pothosware/SoapySDRPlay3 #74: Compile error (undefined reference)

helioshk, 2023-06-30:
Hi, can someone help with that:

[ 16%] Building CXX object CMakeFiles/sdrPlaySupport.dir/Registration.cpp.o
[ 33%] Building CXX object CMakeFiles/sdrPlaySupport.dir/sdrplay_api.cpp.o
[ 50%] Building CXX object CMakeFiles/sdrPlaySupport.dir/Settings.cpp.o
[ 66%] Building CXX object CMakeFiles/sdrPlaySupport.dir/Streaming.cpp.o
[ 83%] Building CXX object CMakeFiles/sdrPlaySupport.dir/Version.cpp.o
[100%] Linking CXX shared module libsdrPlaySupport.so
/usr/bin/ld: CMakeFiles/sdrPlaySupport.dir/Settings.cpp.o:(.data.rel.ro._ZTV12SoapySDRPlay[_ZTV12SoapySDRPlay]+0x338): undefined reference to `SoapySDR::Device::getSettingInfo(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: CMakeFiles/sdrPlaySupport.dir/Settings.cpp.o:(.data.rel.ro._ZTV12SoapySDRPlay[_ZTV12SoapySDRPlay]+0x358): undefined reference to `SoapySDR::Device::getSettingInfo(int, unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/sdrPlaySupport.dir/build.make:163: libsdrPlaySupport.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/sdrPlaySupport.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

fventuri, 2023-06-30:
@helioshk - thanks for submitting this issue.

I just built SoapySDR and SoapySDRPlay3 from the latest version in the `master`  branch, and both builds ran without errors on my Linux desktop (I am on Fedora Core).

Do you mind rerunning the `make` command with the argument `VERBOSE=1`, i.e.:
```
make VERBOSE=1
```
and post the output here, so we can see more details about the failing link command?

Thanks,
Franco

helioshk, 2023-06-30:
@fventuri 

Note that i'm using the SoapySDR library from debian bookworm, might that be an issue?

/usr/bin/cmake -E cmake_link_script CMakeFiles/sdrPlaySupport.dir/link.txt --verbose=1
/usr/bin/c++ -fPIC -O3 -DNDEBUG -pthread -shared  -o libsdrPlaySupport.so CMakeFiles/sdrPlaySupport.dir/Registration.cpp.o CMakeFiles/sdrPlaySupport.dir/sdrplay_api.cpp.o CMakeFiles/sdrPlaySupport.dir/Settings.cpp.o CMakeFiles/sdrPlaySupport.dir/Streaming.cpp.o CMakeFiles/sdrPlaySupport.dir/Version.cpp.o  -Wl,-rpath,/usr/local/lib: /usr/local/lib/libsdrplay_api.so /usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.8.1 -Wl,--no-undefined -pthread 
/usr/bin/ld: CMakeFiles/sdrPlaySupport.dir/Settings.cpp.o:(.data.rel.ro._ZTV12SoapySDRPlay[_ZTV12SoapySDRPlay]+0x338): undefined reference to `SoapySDR::Device::getSettingInfo(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: CMakeFiles/sdrPlaySupport.dir/Settings.cpp.o:(.data.rel.ro._ZTV12SoapySDRPlay[_ZTV12SoapySDRPlay]+0x358): undefined reference to `SoapySDR::Device::getSettingInfo(int, unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/sdrPlaySupport.dir/build.make:163: libsdrPlaySupport.so] Error 1
make[2]: Leaving directory '/home/helios/git/SoapySDRPlay/build'
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/sdrPlaySupport.dir/all] Error 2
make[1]: Leaving directory '/home/helios/git/SoapySDRPlay/build'

helioshk, 2023-06-30:
My bad, i found remnants of an old soapysdr installation in /usr/local
I removed them and now it compiles.

closed 2023-06-30 by helioshk