pothosware/SoapySDRPlay3 #75: Is HDR mode supported by SoapySDRPlay3?

oh2lak, 2023-07-09:
SDRPlay devices like RSPDx supports HDR mode (High Dynamic Range) in conjunction with 500kHz and 2MHz low-pass filters. Is this feature supported by SoapySDRPlay3 or not?

As the bands below 2MHz are completely dead with RSPDx being used with OpenWebRX, I think lack of support for the LPF filters or HDR mode is the reason for that.

Usage of the 'hdrEnable' command is described in the SDRPlay API document;
https://www.sdrplay.com/docs/SDRplay_API_Specification_v3.07.pdf#page=26

fventuri, 2023-07-10:
@oh2lak - a few months ago I created a branch called `rspdx-hdr` (https://github.com/pothosware/SoapySDRPlay3/tree/rspdx-hdr), where I added a new boolean setting called `hdr_ctrl` (https://github.com/pothosware/SoapySDRPlay3/blob/rspdx-hdr/Settings.cpp#L1386-L1392).

After you build and install that branch, you can enable HDR mode by adding `hdrl_ctrl=1` to the device string.

Please give it a try and let me know what you think; once we think it is good and works as expexted, I'll merge that change into the `master` branch.

Franco

oh2lak, 2023-07-10:
Thanks Franco, I will give it a try! Looks like without the HDR setting enabled, RSPDx uses the lowest BPF which is 2-12MHz, thus making it unusable for LW and MW use.

I'll report back how it goes!

Erik

oh2lak, 2023-07-21:
I can confirm that the `rspdx-hdr` branch works with HDR mode and I have signals on MW and LW now. Though, I did not include the parameter `hdrl_ctrl=1` anywhere. I'm testing the driver with OpenWebRX

..
Erik

fventuri, 2023-07-25:
Erik,
thanks for confirming that that change works.
I just merged that branch into the `master` branch and updated the release version to 0.4.2.

I think the reason why HDR mode is enabled in your RSPdx is because the default for that setting is `true` (https://github.com/pothosware/SoapySDRPlay3/blob/master/Settings.cpp#L1394); you can disable it by adding `hdr_ctrl=0` in the device configuration string.

Franco