pothosware/SoapySDRPlay3 #77: Confusing partial support for rspduo in Dual Tuner mode

isomer, 2023-09-03:
I was confused trying to use Dual Tuner mode for an rspduo, and spent a few evenings trying to get it to work before I read the code and realised that it's not actually supported.

The driver [registers](https://github.com/pothosware/SoapySDRPlay3/blob/master/Registration.cpp#L100-L111 ) a "dual tuner" mode which I had assumed meant that the driver supported this, however the driver [rejects all](https://github.com/pothosware/SoapySDRPlay3/blob/master/Streaming.cpp#L240-L243) setup requests with multiple channels (since it's [not supported](https://github.com/pothosware/SoapySDRPlay3/blob/master/Streaming.cpp#L440-L441)).

This confused me for some time.
 - Ideally, there would be support added for dual tuners (there's some code in issue #17 that attempts to do this).
 - The error message could be improved, instead of "setupStream invalid channel selection" (which made me think I was specifying the channels incorrectly, eg perhaps they were 1 and 2, instead of 0 and 1?), something like "setupStream multiple channels not supported, or channel out of range" would have made that clear what was wrong.
 -  Why even register a mode that the driver doesn't actually properly support?  Perhaps https://github.com/pothosware/SoapySDRPlay3/blob/master/Registration.cpp#L100-L111 should be #ifdef'd out until support is added to avoid confusion?  People can use the master/slave versions instead?

fventuri, 2023-09-04:
First of all my apologies for the confusion and the frustration in using this SoapySDR module with your RSPduo in Dual Tuner - to be honest not many people use SoapySDRPlay3  with an RSPduo in Dual Tuner mode, so there are probably some rough edges.

Anyway tonight I just wrote a simple C++ program based on SoapySDR API C++ example code (https://github.com/pothosware/SoapySDR/wiki/Cpp_API_Example), added some code to handle the dual tuner case (since this was just for a quick test I did a lot of cut&paste, and the code doesn't look very elegant), compiled, ran a quick test with my RSPduo, and I see samples being received from both tuners:
```
[INFO] devIdx: 0
[INFO] SerNo:
[INFO] hwVer: 3
[INFO] rspDuoMode: 2
[INFO] tuner: 3
[INFO] rspDuoSampleFreq: 6000000.000000
Rx antennas: Tuner 1 50 ohm,
Rx Gains: IFGR, RFGR, 
Rx freq ranges: [1000 Hz -> 2e+09 Hz], 
Num channels: 2
[INFO] Using format CF32.
[INFO] Using format CF32.
elapsed time = 10012307 µs
nSamples0 = 19919104
nSamples1 = 19919104
Done
```

The code is attached; please take a look and give it a try.

You may want also to take a look at this discussion from 2-3 years ago (https://github.com/pothosware/SoapySDRPlay3/issues/12) about the SDR program SdrGlut, and some considerations when using the SoapySDRPlay3 module in Dual Tuner mode.

Franco

[ExampleDualTuner.zip](https://github.com/pothosware/SoapySDRPlay3/files/12508859/ExampleDualTuner.zip)

isomer, 2023-09-04:
Oh interesting.  I was expecting it to have one stream with two channels, which is what was confusing me.  I didn't realise you could open two streams from the same SDR.  (although looking at it now it makes sense).

Can I suggest you create an examples/ directory in the repository and put this there so that future explorers can easily find it?  It demonstrates a bunch of useful features (not just dual tuner mode).