pothosware/SoapySDRPlay3 #78: Gqrx failed to recognize device due to ABI breakage

EL-File4138, 2023-09-17:
Fedora 38 with Gqrx install by dnf (It's dependencies will require SoapySDR).
```
Installed SoapySDR-0.8.1-8.fc38.x86_64                        @fedora
Installed gnuradio-3.10.4.0-6.fc38.x86_64                     @fedora
Installed gr-funcube-1.0.0-19.20220130gitbf71b979.fc38.x86_64 @fedora
Installed gr-iqbal-0.38.2-19.20210210gitfbee239a.fc38.x86_64  @fedora
Installed gr-osmosdr-0.2.4-3.fc38.x86_64                      @fedora
Installed python3-gnuradio-3.10.4.0-6.fc38.x86_64             @fedora
Installed gqrx-2.16-1.fc38.x86_64                             @updates
```
Built and installed by the method in Wiki. Shared library built has an ABI of `0.8-3`, while the SoapySDR has `0.8`, thus not installed in the correct folder that SoapySDR will recognize. Manually moved the library file, `SoapySDRUtils` recognized it and the device:
```
> SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-gbb33b2d2
API Version: v0.8.200
ABI Version: v0.8-3
Install root: /usr/local
Search path:  /usr/local/lib64/SoapySDR/modules0.8-3
Module found: /usr/local/lib64/SoapySDR/modules0.8-3/libsdrPlaySupport.so (0.4.1-534a173)
Available factories... sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

```
>SoapySDRUtil --probe="driver=sdrplay"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device device=sdrplay
[INFO] devIdx: 0
[INFO] SerNo: <Redacted>
[INFO] hwVer: 1
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=RSP1
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
...
```

But when launching Gqrx, it shows:
```
[ERROR] SoapySDR::loadModule(/usr/local/lib64/SoapySDR/modules0.8/libsdrPlaySupport.so)
  sdrplay failed ABI check: Library ABI=0.8, Module ABI=0.8-3
```

and thus not recognizing the device.
Is there anything I've done wrong?

fventuri, 2023-09-17:
@EL-File4138 - I am running Fedora 38 here, however in my case I chose not to install any of the SDR related packages from the distribution, but I built everything from source.

In your case I suspect there's some overlap between the distribution packages (which typically end up under `/usr`) and what you built and installed from source (which you would find under `/usr/local`); if I understand correctly you have two versions of SoapySDR, one installed from the Fedora package and the other one built from source.

If this is the case I would remove the version installed from source (to avoid conflicts), i.e. I would run something like `rm /usr/local/ib64/libSoapySDR.* /usr/local/lib64/SoapySDR/modules0.8/* /usr/local/include/SoapySDR*` and so on), install the Fedora package `SoapySDR-devel`, and finally rebuild this module from source following the instructions here: https://github.com/pothosware/SoapySDRPlay3/wiki#building-soapy-sdr-play

Franco

EL-File4138, 2023-09-18:
@fventuri This is indeed the problem. The source build one comes from the driver install script of SDRplay, and I didn't notice that. Thanks for pointing out!

closed 2023-09-18 by EL-File4138