pothosware/SoapySDRPlay3 #80: How to set the antenna to use?

BikerJohn66, 2024-01-07:
For the dumpHFDL program I use a SDRdx via SoapySDR on Raspberry Pi 4 that's running the bookworm operating system.
The SoapySDRUtil --probe"driver=sdrplay" result is:

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay
[INFO] devIdx: 0
[INFO] SerNo: [removed]
[INFO] hwVer: 4
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=RSPdx
  sdrplay_api_api_version=3.120000
  sdrplay_api_hw_version=4

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]
     * DabNotch Enable - DAB Notch Filter Control
       [key=dabnotch_ctrl, default=true, type=bool]
     * HDR Enable - RSPdx HDR Control
       [key=hdr_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: Antenna A, Antenna B, Antenna C
  Corrections: DC removal
  Full gain range: [0, 66] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 27] dB
  Full freq range: [0.001, 2000] MHz
    RF freq range: [0.001, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.0625, 0.096, 0.125, 0.192, 0.25, ..., 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

How can I set the "Antenna C"  as the one to be used? I tried all variations I could think of of driver=sdrplay,antenna="Atenna C" but the result is the same:

driver=sdrplay,ant=Antenna C: center frequency set to 3814.000 kHz
driver=sdrplay,ant=Antenna C: frequency correction set to 0.00 ppm
driver=sdrplay,ant=Antenna C: auto gain enabled
driver=sdrplay,ant=Antenna C: using antenna Antenna A
driver=sdrplay,ant=Antenna C: using native sample format CS16 (full_scale: 32767.000)

Thanks in advance!

zuckschwerdt, 2024-01-07:
Isn't the antenna arg `antenna` and not `ant`? If you found `ant` mentioned somewhere that should be corrected.

BikerJohn66, 2024-01-07:
Just found it by accident. driver=sdrplay --antenna="Antenna C".
Thank you very much for your help!

closed 2024-01-07 by BikerJohn66