pothosware/SoapySDRPlay3 #83: build error

denk1, 2024-03-30:
I've got these errors on a Raspberry Pi4 
error: ‘struct sdrplay_api_DeviceT’ has no member named ‘valid’
   52 |       if (not rspDevs[i].valid) continue;

error: ‘SDRPLAY_RSP1B_ID’ was not declared in this scope; did you mean ‘SDRPLAY_RSP1_ID’?
   66 |       else if (rspDevs[i].hwVer == SDRPLAY_RSP1B_ID)
      |                                    ^~~~~~~~~~~~~~~~

Ho can I resolve the problem?

fventuri, 2024-03-30:
@denk1 - those error messages are typically a symptom that you have the old version of the SDRplay API on your computer.
You can find the installation file for the latest version (3.14 as of writing) of the SDRplay API here: https://www.sdrplay.com/api/

Franco

denk1, 2024-03-31:
@fventuri but I have the 32 bit architecture. Which the latest version of SoapySDRPlay could be compatible with this one?

fventuri, 2024-03-31:
@denk1 - in that case you have to use the code from the `api-3.07` branch, i.e.:
```
git clone https://github.com/pothosware/SoapySDRPlay3.git
cd SoapySDRPlay3
git checkout api-3.07
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig
```

Also Raspberry Pi now offers a 64 bit version of Raspberry Pi OS (I think it is the recommended version now).
You may want to consider switching to it since SDRplay API version 3.07 is now deprecated, and I don't plan to make any future changes, improvements, or fixes to the `api-3.07` branch.

Franco

denk1, 2024-03-31:
@fventuri thanks a lot for the information but the problem is I have already used this version for a long time and it doesn't work well. I always get this errors:
```
[ 1452.483388] usb 1-1-port2: attempt power cycle
[ 1453.651435] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[ 1454.503484] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
```
and my driver gets failed 
I hoped that problem was resolved in the latest version, however, it  isn't compatible with Raspberry Pi 4 due to its old architecture 
My cable is perfect one I've tried using many different ones and it has the same issue

fventuri, 2024-03-31:
@denk1 - I just looked at your error messages, and I suspect your problem is not in the driver or the version of the SDRplay API you are using.

Can you try to plug your SDR to a different Linux computer (or a different Raspberry Pi) to see if you still get the same error messages in 'dmesg' in that second computer you connected it to?

Also can you run this command:
```
lsusb -d 1df7:
```
to see if it shows your RSP?

Franco

denk1, 2024-04-01:
@fventuri after I get those errors that dmesg showing 
my devices in `lsusb -d 1df7:` stop being seen

denk1, 2024-04-01:
@fventuri it works well on x86_64/. I will try to plug to  a different Raspberry Pi device. Thanks.

closed 2024-04-01 by denk1