pothosware/SoapySDRPlay3 #87: Random crash with rspduo

balthus, 2024-05-08:
Hello,
My setup used to work fine with previous ubuntu version (18.x)
I migrate to ubuntu 22.04 few month ago.
Recently I tried to make it work again by recompiling soapysdrplay and use gqrx with soapy driver.
It works fine for a while but after some random time, it suddenly crash.
It seems to comes from soapySdrplay layer or something wrong with libusb.
The service crash and restart. Sometimes I even have to restart the service after a crash:

**journalctl -u sdrplay -f**
```
May 08 17:58:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: ReadPacket: Gain update confirmed: Gr=39dB GrToggle=1 gset=0x227 tunerA
May 08 17:59:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: streamThread: M: 37142820: iDCoffset=793.659429752 qDCoffset=0.000000000
May 08 17:59:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: setGr: Tuner1: in: 39 1 0
May 08 17:59:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x09, 0x012271)
May 08 17:59:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: setGr: Tuner1: GR->39[39,7,0,0] gRset->0x227 DCCALmode=4 DCCALspd=0 GrToggle->0 (1 0)
May 08 17:59:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: GetCurrentGainVals: hwVer=2 freq=445.46 bw=1536 if=1620, bbGr=39, lnaState=1, minGr=20, amPort1En=0, apiDownConvert=1
May 08 17:59:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: GetCurrentGainVals: hwVer=2 freq=445.46 group=8 gainState=1, corr=-9.50, freq0=420.00 freq1=480.00 gain0=116.19 gain1=117.13: curr_no_bb=107.09 max=94.51 min=15.53 curr=68.09
May 08 17:59:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: setGr: Tuner1: OUT 39 46 1
May 08 17:59:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: ReadPacket: Gain update confirmed: Gr=39dB GrToggle=0 gset=0x227 tunerA
May 08 18:00:06 xxx-desktop sdrplay_apiService[1171897]: libusb: error [submit_iso_transfer] submiturb failed, errno=12
May 08 18:00:06 xxx-desktop sdrplay_apiService[1171897]: [1172136]: devIdx0: sdrplay_apiService_rsp: libusbDriverCallback: Error: libusb_submit_transfer(0x7f7c780017d0) -1
May 08 18:00:06 xxx-desktop sdrplay_apiService[1171897]: libusb: error [submit_iso_transfer] submiturb failed, errno=12
May 08 18:00:06 xxx-desktop sdrplay_apiService[1171897]: [1172136]: devIdx0: sdrplay_apiService_rsp: libusbDriverCallback: Error: libusb_submit_transfer(0x7f7c78001f00) -1
May 08 18:00:06 xxx-desktop sdrplay_apiService[1171897]: libusb: error [submit_iso_transfer] submiturb failed, errno=12
May 08 18:00:06 xxx-desktop sdrplay_apiService[1171897]: [1172136]: devIdx0: sdrplay_apiService_rsp: libusbDriverCallback: Error: libusb_submit_transfer(0x7f7c7800b100) -1
May 08 18:00:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: streamThread: M: 37857105: iDCoffset=796.094024778 qDCoffset=0.000000000
May 08 18:00:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: setGr: Tuner1: in: 39 1 0
May 08 18:00:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x09, 0x012271)
May 08 18:00:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: setGr: Tuner1: GR->39[39,7,0,0] gRset->0x227 DCCALmode=4 DCCALspd=0 GrToggle->1 (1 0)
May 08 18:00:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: GetCurrentGainVals: hwVer=2 freq=445.46 bw=1536 if=1620, bbGr=39, lnaState=1, minGr=20, amPort1En=0, apiDownConvert=1
May 08 18:00:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: GetCurrentGainVals: hwVer=2 freq=445.46 group=8 gainState=1, corr=-9.50, freq0=420.00 freq1=480.00 gain0=116.19 gain1=117.13: curr_no_bb=107.09 max=94.51 min=15.53 curr=68.09
May 08 18:00:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: setGr: Tuner1: OUT 39 46 1
May 08 18:00:14 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: ReadPacket: Gain update confirmed: Gr=39dB GrToggle=1 gset=0x227 tunerA
May 08 18:00:20 xxx-desktop sdrplay_apiService[1171897]: libusb: error [submit_iso_transfer] submiturb failed, errno=12
May 08 18:00:20 xxx-desktop sdrplay_apiService[1171897]: [1172136]: devIdx0: sdrplay_apiService_rsp: libusbDriverCallback: Error: libusb_submit_transfer(0x7f7c7800b850) -1
May 08 18:00:27 xxx-desktop sdrplay_apiService[1171897]: libusb: error [submit_iso_transfer] submiturb failed, errno=12
May 08 18:00:27 xxx-desktop sdrplay_apiService[1171897]: [1172136]: devIdx0: sdrplay_apiService_rsp: libusbDriverCallback: Error: libusb_submit_transfer(0x7f7c78000980) -1
May 08 18:00:34 xxx-desktop sdrplay_apiService[1171897]: libusb: error [submit_iso_transfer] submiturb failed, errno=12
May 08 18:00:34 xxx-desktop sdrplay_apiService[1171897]: [1172136]: devIdx0: sdrplay_apiService_rsp: libusbDriverCallback: Error: libusb_submit_transfer(0x7f7c7800bfa0) -1
May 08 18:00:48 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: streamThread: Gap=0x2b20 e=0x7d5d1dd2 a=0x7d5d48f2
May 08 18:00:54 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: streamThread: Gap=0x78c e=0x7fa8b9e2 a=0x7fa8c16e
May 08 18:00:56 xxx-desktop sdrplay_apiService[1171897]: libusb: error [submit_iso_transfer] submiturb failed, errno=12
May 08 18:00:56 xxx-desktop sdrplay_apiService[1171897]: [1172136]: devIdx0: sdrplay_apiService_rsp: libusbDriverCallback: Error: libusb_submit_transfer(0x7f7c7800cc30) -1
May 08 18:00:56 xxx-desktop sdrplay_apiService[1171897]: libusb: error [submit_iso_transfer] submiturb failed, errno=12
May 08 18:00:56 xxx-desktop sdrplay_apiService[1171897]: [1172136]: devIdx0: sdrplay_apiService_rsp: libusbDriverCallback: Error: libusb_submit_transfer(0x7f7c78006440) -1
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: usb_StreamRead: usb_StreamRead: Read failed to read sufficient data: requested 393216 read 384000
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: bridge_StreamRead: bridge_StreamRead: Read failed 1
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: libusb: error [submit_iso_transfer] submiturb failed, errno=12
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: usb_StreamInit: Error: libusb_submit_transfer() -1
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: bridge_StreamInit: bridge_StreamInit: Open failed
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: bridge_StreamRead: bridge_StreamRead: Read failed 1
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: libusb: error [submit_iso_transfer] submiturb failed, errno=12
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: usb_StreamInit: Error: libusb_submit_transfer() -1
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: bridge_StreamInit: bridge_StreamInit: Open failed
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: bridge_StreamRead: bridge_StreamRead: Read failed 1
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: libusb: error [submit_iso_transfer] submiturb failed, errno=12
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: usb_StreamInit: Error: libusb_submit_transfer() -1
May 08 18:00:57 xxx-desktop sdrplay_apiService[1171897]: [1182166]: devIdx0: sdrplay_apiService_rsp: bridge_StreamInit: bridge_StreamInit: Open failed
May 08 23:43:21 xxx-desktop sdrplay_apiService[1171897]: [1182165]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x09, 0x00001c)
May 08 23:43:21 xxx-desktop sdrplay_apiService[1171897]: [1182165]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x09, 0x000000)
[... cut ...]
I had an usb extension cable and thought it could be faulty, so I retried with a smaller shielded cable but it still crash.
*It used to work fine for hours with previous ubuntu version* and some gqrx binaries I found that probably came with some sdrplay driver.
I also tried to different usb port
I don't know how I can troubleshoot further this issue.
Additional info:
```
uname -a
Linux xxx-desktop 5.15.0-105-generic #115-Ubuntu SMP Mon Apr 15 09:52:04 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux`

lsusb
Bus 003 Device 003: ID 1df7:3010 SDRplay RSP2/RSP2pro
...
```

fventuri, 2024-05-08:
@balthus - thanks for reporting the problem.

Do you mind running the command:
```
sudo dmesg
```
when this problem is occurring to see if there are USB warning or errors in that log?

Also can you run the command:
```
SoapySDRUtil --args="driver=sdrplay" --rate=2e6 --direction=RX
```
which will continuously stream from your RSPduo, and let it run for a while to see if it crashes with the same error?

Franco

balthus, 2024-05-09:
Hello Franco, 
Thanks for your quick reply and support. I think the issue is probably in the sdrplay driver, but I didn't find so far people with similar issue.
I just tried your first command and let it run for a while , then it suddenly get stuck:
```
SoapySDRUtil --args="driver=sdrplay" --rate=2e6 --direction=RX |tee /tmp/soapySDRUtil.log
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] devIdx: 0
[INFO] SerNo: 1808068C11
[INFO] hwVer: 2
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
[INFO] Using format CS16.
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin RX rate test at 2 Msps
Starting stream loop, press Ctrl+C to exit...
1.88224 Msps	7.52897 MBps
1.94331 Msps	7.77324 MBps
1.95985 Msps	7.83938 MBps
1.971 Msps	7.88398 MBps
1.97769 Msps	7.91074 MBps
...
1.99939 Msps	7.99755 MBps
1.99935 Msps	7.99741 MBps
-
```
Looking at dmesg, there is no event when the error occured:
```
dmesg -T
[Thu May  9 00:42:43 2024] audit: type=1107 audit(1715208163.482:528): pid=1178 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.20" mask="receive" pid=8485 label="snap.firefox.firefox" peer_pid=1220 peer_label="unconfined"
                            exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[Thu May  9 11:33:53 2024] audit: type=1107 audit(1715247233.834:529): pid=1178 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.20" mask="receive" pid=8485 label="snap.firefox.firefox" peer_pid=1220 peer_label="unconfined"
                            exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'

```
```
May 09 11:56:17 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: setGr: Tuner1: in: 36 0 0
May 09 11:56:17 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x09, 0x010241)
May 09 11:56:17 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: setGr: Tuner1: GR->36[36,0,0,0] gRset->0x24 DCCALmode=4 DCCALspd=0 GrToggle->1 (1 0)
May 09 11:56:17 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: GetCurrentGainVals: hwVer=2 freq=200.00 bw=1536 if=1620, bbGr=36, lnaState=0, minGr=20, amPort1En=0, apiDownConvert=1
May 09 11:56:17 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: GetCurrentGainVals: hwVer=2 freq=200.00 group=4 gainState=0, corr=-9.50, freq0=198.00 freq1=224.00 gain0=127.16 gain1=126.22: curr_no_bb=117.59 max=97.59 min=-3.04 curr=81.59
May 09 11:56:17 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: setGr: Tuner1: OUT 36 36 0
May 09 11:56:18 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: ReadPacket: Gain update confirmed: Gr=36dB GrToggle=1 gset=0x24 tunerA
May 09 11:56:18 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: streamThread: master grChanged @ 7567533
May 09 11:56:18 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: dcOffsetCorrection: Tuner1: dcOffsetCorrection: switched to tracking mode diff=39 curr=687.62 last=726.85
May 09 11:58:55 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: setGr: Tuner1: in: 34 0 0
May 09 11:58:55 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x09, 0x010221)
May 09 11:58:55 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: setGr: Tuner1: GR->34[34,0,0,0] gRset->0x22 DCCALmode=4 DCCALspd=0 GrToggle->0 (1 0)
May 09 11:58:55 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: GetCurrentGainVals: hwVer=2 freq=200.00 bw=1536 if=1620, bbGr=34, lnaState=0, minGr=20, amPort1En=0, apiDownConvert=1
May 09 11:58:55 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: GetCurrentGainVals: hwVer=2 freq=200.00 group=4 gainState=0, corr=-9.50, freq0=198.00 freq1=224.00 gain0=127.16 gain1=126.22: curr_no_bb=117.59 max=97.59 min=-3.04 curr=83.59
May 09 11:58:55 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: setGr: Tuner1: OUT 34 34 0
May 09 11:58:55 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: ReadPacket: Gain update confirmed: Gr=34dB GrToggle=0 gset=0x22 tunerA
May 09 11:58:55 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: streamThread: master grChanged @ 9440304
May 09 11:58:55 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: dcOffsetCorrection: Tuner1: dcOffsetCorrection: switched to tracking mode diff=45 curr=787.09 last=741.24
[... cut ...]
May 09 12:01:15 xxx-desktop-desktop sdrplay_apiService[1252435]: libusb: error [submit_iso_transfer] submiturb failed, errno=12
May 09 12:01:15 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: usb_StreamInit: Error: libusb_submit_transfer() -1
May 09 12:01:15 xxx-desktop-desktop sdrplay_apiService[1252435]: [1356766]: devIdx0: sdrplay_apiService_rsp: bridge_StreamInit: bridge_StreamInit: Open failed
```
I think the issue is in the sdrplay driver not in the SoapySDRPlay3 but I don't know what else I can try or check...
Best regards,

alphafox02, 2024-05-09:
If you have the time (I’m just really curious) could you boot this ISO live after burning to USB 
https://sourceforge.net/projects/dragonos-focal/files/
Plug in your sdrplay without extension cable and perform same commands/use Gqrx etc. Everything is already included.

I have several pieces of SDRPlay equipment here and I will try to reproduce as well, so far I’m unable.

SDRplay, 2024-05-09:
It looks like your system is showing a lot of Isochronous transfer errors. You could try bulk mode. Build SoapySDRPlay library with -DSTREAMING_USB_MODE_BULK and try again. I would also try without debug enabled as that will have quite a performance impact.

balthus, 2024-05-09:
Thanks for your suggestions, I will try all of them coz it's annoying that it crashes after 10 or 15 min ;(
for the DSTREAMING_USB_MODE_BULK, should i recompile like
```
cmake -DSTREAMING_USB_MODE_BULK=true ..
make
sudo make install
```
I'm not so familiar with C compilation  (too many years ;)) Is there some log or way to check it was correctly compiled with this settings? I want to be sure it's correctly done.

I will keep you updated with results. Many thanks !

balthus, 2024-05-09:
Ok, it seems bulk mode enabled because of the gqrx output:
```
gqrx
WARNING: CPU random generator seem to be failing, disabling hardware random number generation
WARNING: RDRND generated: 0xffffffff 0xffffffff 0xffffffff 0xffffffff
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file fcd rtl rtl_tcp hackrf rfspace soapy redpitaya 
file_source :warning: file size is not a multiple of item size
rx_nb_cc :info: set_min_output_buffer on block 12 to 32768
Resampling audio 96000 -> 48000
BandPlanFile is /home/balthus/.config/gqrx/bandplan.csv
BookmarksFile is /home/balthus/.config/gqrx/bookmarks.csv
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file fcd rtl rtl_tcp hackrf rfspace soapy redpitaya 
[INFO] devIdx: 0
[INFO] SerNo: 1808068C11
[INFO] hwVer: 2
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
[INFO] Using format CF32.
[WARNING] Not updating IFGR gain because AGC is enabled
[WARNING] Not updating IFGR gain because AGC is enabled
[INFO] Using streaming USB mode bulk.
```
**[INFO] Using streaming USB mode bulk.**
Will launch some tests and see how it goes :)

balthus, 2024-05-09:
Hi, 
Many thanks for the -DSTREAMING_USB_MODE_BULK
I let gqrx to run in the background for more than 3 hours and no crash!
I don't know why the default way doesn't work. The pc is recent and quite powerful (AMD Ryzen 5 3600)

Anyway I'm happy that it doesn't crash any more. Thanks again!
You can close the issue :)

closed 2024-05-09 by balthus