pothosware/SoapySDRPlay3 #90: RSPDuo can't disable agc

f4tnk, 2024-10-06:
Hi,

When I use SoapySDRplay on with my RSPDUo, I cannot turn off the AGC so that I can adjust the IFGR and RFGR gain.

I tried driver=sdrplay,agc_setpoint=0 but it doesn't work and I get the following message: 

[WARNING] Not updating IFGR gain because AGC is enabled

Is there something I'm doing wrong?

Thank you in advance for your help

F4TNK

fventuri, 2024-10-07:
@f4tnk - the SoapySDR API provides the method `setGainMode()` (https://github.com/pothosware/SoapySDR/blob/master/include/SoapySDR/Device.hpp#L702-L708) to enable/disable AGC.

I am not sure what programming language your application is written in, but you can see here how CubicSDR uses that method in C++: https://github.com/cjcliffe/CubicSDR/blob/master/src/sdr/SoapySDRThread.cpp#L129-L131

Python has a very similar way of setting it - this is how you can disable AGC in Python:
```
    sdr.setGainMode(SOAPY_SDR_RX, 0, False)
```
The Python code above should work with the basic Python example you can find here: https://github.com/pothosware/SoapySDR/wiki/PythonSupport#basic-example

73,
Franco K4VZ

Franco

f4tnk, 2024-10-07:
Hello Franco,

Thank you for your response and information.

In fact I use the satgnos suite for satellite tracking and decoding.

When I use CubicSDR, I can deactivate the AGC and adjust both gains... so it's not a SoapySDRPlay3 problem when you think about it.

There must be a problem elsewhere in the chain... I'll take a look at gr-soapy and the other modules.

Thank you again and good luck.

73's F4TNK

closed 2024-10-07 by f4tnk