pothosware/SoapySDRPlay3 #91: cmake fails with the following error

antzi, 2025-04-26:
```
antzi@satnogs:~/sw/SoapySDRPlay/build $ cmake ..
-- The CXX compiler identification is GNU 12.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:7 (find_package):
  Could not find a package configuration file provided by "SoapySDR"
  (requested version 0.4.0) with any of the following names:

    SoapySDRConfig.cmake
    soapysdr-config.cmake

  Add the installation prefix of "SoapySDR" to CMAKE_PREFIX_PATH or set
  "SoapySDR_DIR" to a directory containing one of the above files.  If
  "SoapySDR" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/antzi/sw/SoapySDRPlay/build/CMakeFiles/CMakeOutput.log".
```

PFA the above log: [CMakeOutput.log](https://github.com/user-attachments/files/19923907/CMakeOutput.log)

Building on Raspberry Pi 4B, Linux satnogs 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux

SDRplay API installed - SDRplay_RSP_API-Linux-3.15.2.run

zuckschwerdt, 2025-04-26:
If you want SoapySDR support you'll need `libsoapysdr-dev`

antzi, 2025-04-26:
After installing the `libsoapysdr-dev`, the building completed without error. Thanks!