pothosware/SoapySDRPlay3 #92: Dual Tuner Usage with Soapy (rtl433)

alphafox02, 2025-05-14:
Maybe a long shot, but I thought it would be best to ask here for assistance. I’ve been able to use SDRPlay hardware with Soapy and Rtl433. What I’ve yet to understand is how to take advantage of SDRPlay’s dual tuner SDR in such a way that would allow me to run two instances of rtl433 on each tuner on different frequencies. 

In CubicSDR I’m able to accomplish this, but I’m not fully understanding the proper device strings to pass into each instance of rtl433 so that the radio is “initiated” in dual tuner mode, i think is what I’m trying to say. 

Any chance someone has done this or has a minute to try it themselves? I’d like to tune to weather devices and other stuff at the same time using the one SDRPlay SDR. 

https://github.com/merbanan/rtl_433

fventuri, 2025-05-14:
@alphafox02 - if I understand correctly, you plan to use the two RSPduo tuners in different applications.
You can do this by running your RSPduo in master/slave mode.

One of your applications (the 'main' one) will be connecting to the RSPduo in master mode, and will control most of the common settings of the RSPduo (sample rate, IF frequency, and so on), while the other one will be running in 'slave' mode and will be able to control only a limited number of settings (like center frequency, gains, etc).

In the `SoapySDRPlay3` driver, you can select these RSPduo modes of operation by using the `mode` parameter in the device string, for instance:
```
# master mode
mode=MA
```
or
```
# slave mode
mode=SL
```

See:
- https://github.com/pothosware/SoapySDRPlay3/blob/master/Settings.cpp#L49
- https://github.com/pothosware/SoapySDRPlay3/blob/master/Settings.cpp#L1948
- https://github.com/pothosware/SoapySDRPlay3/blob/master/Settings.cpp#L1960

Franco

alphafox02, 2025-05-14:
Thank you! As soon as I’m back with the RSPDuo I will try and pass device strings as you mention and see if I can get two instances of RTL433 up and running.

fventuri, 2025-05-21:
I am glad to hear master/slave mode works for you.

If you already have the IF AGC turned on, and you still see ADC overloads, you could try increasing the RF gain reduction (aka LNA state) by increasing the value of `rfgain_sel` (https://github.com/pothosware/SoapySDRPlay3/blob/master/Settings.cpp#L1590) until you're happy with the result.

Every increment of LNA state via `rfgain_sel` adds more attenuation in the RF stage. How much extra attenuation depends on the RSP model, the band you are on, and in some cases the antenna input. You can find all the details in the 'Gain Reduction Tables' section of the SDRplay API specification guide: https://www.sdrplay.com/docs/SDRplay_API_Specification_v3.15.pdf

Franco